freifunk-berlin/bbb-configs #4: Implement Core Router to act NTP Server and make aps to use it.

spolack, 2021-01-28:


Noki, 2022-06-20:
NTP server is enabled for core routers and uplink gateways since https://github.com/freifunk-berlin/bbb-configs/commit/53ae1611ff3d334bede50490d0c965edc7ae4db5 now only the usage by the APs is missing.

Noki, 2022-10-02:
This can be done the same way setting dns and gateway work, by introducing a new field "ntp" in network.yml that can be read like this: https://github.com/freifunk-berlin/bbb-configs/blob/e706067ac1b0055d32057c8b4c2ebe5056822252/roles/cfg_openwrt/templates/common/config/network.j2#L43-L48

Besides that the following must be different for APs: https://github.com/freifunk-berlin/bbb-configs/blob/53ae1611ff3d334bede50490d0c965edc7ae4db5/roles/cfg_openwrt/templates/common/config/system.j2#L21-L23

PR #340 (2022-12-03): Make APs use NTP server of the core router
This pull request should address the remaining stuff to close #4.

To actually start using the NTP server on the core router the corresponding networks files have to be changed later on.
closed 2022-12-03 by Akira25