freifunk-berlin/bbb-configs #9: Implement DSA switching

spolack, 2021-02-17:
Should be quite Easy:
```
config device
  option type bridge
  option name switch0

config bridge-vlan
    option vlan 'VLANID'
    list ports 'lan1:t'
    list ports 'lan2:t'
    list ports 'lan3:t'
    list ports 'lan4:t'

config interface NETWORK
  # For the time being we do need the redundant bridge
  option type bridge
  option ifname switch0.VLANID
```

PolynomialDivision, 2021-02-17:
We have now the situation that some targets have DSA and some have still swconfig, or?

spolack, 2021-02-17:
Yes, but as we anyway have to maintain the Port definitiions per model, we can just set a DSA flag there.

- acually a low brainer.

https://github.com/Freifunk-Spalter/bbb-configs/blob/master/group_vars/model_tplink_archer_c6_v2.yml

spolack, 2021-02-20:
I think i implemented dsa correctly. That would be the resulting config if i would use an ER-X as scharni-core.
```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option dns '46.182.19.48 80.67.169.40 194.150.168.168 2001:4ce8::53 2001:910:800::12'
	option ip6addr '2001:bf7:830:8400::/56'

config device
	option type 'bridge'
	option name 'switch0'

                    
config bridge-vlan
	option vlan '12'
	option ports 'eth0:t eth1:t eth2:t eth3:t eth4:t'
                    
config bridge-vlan
	option vlan '40'
	option ports 'eth0:t eth1:t eth2:t eth3:t eth4:t'
                    
config bridge-vlan
	option vlan '42'
	option ports 'eth0:t eth1:t eth2:t eth3:t eth4:t'

config interface 'mesh_sama'
	option proto 'static'
	option ifname 'switch0.12'
	option ipaddr '10.31.252.193/32'
	option ip6addr '2001:bf7:830:84fe::1/128'

config interface 'dhcp'
	option proto 'static'
	option ifname 'switch0.40'
	option ipaddr '10.31.79.193/26'
	option macaddr '02:00:00:00:00:01'
	option ip6addr '2001:bf7:830:8400::1/64'

config interface 'mgmt'
	option proto 'static'
	option ifname 'switch0.42'
	option ipaddr '10.31.52.161/27'
	option ip6addr '2001:bf7:830:8401::1/64'
```

@pmelange  Unfortunately i've got no ER-X SFP nor other device for testing. Coul you please review and test, if possible?

spolack, 2021-02-21:
Implemented in 23b57e1bfccd80ca331a327ad8c5de2337cf1543  - Special thanks to @pmelange for testing!

closed 2021-02-21 by spolack

pmelange, 2021-02-21:
I'm happy to be able to help