freifunk-berlin/bbb-configs #12: Uplink-Gateways dont announce any prefixes.

spolack, 2021-03-06:
..therefore they are not reachable by their internal ipv6 mesh ips.

PolynomialDivision, 2021-03-06:
btw. I wrote here some tool to automatically get the ip addresses of a gateway:
https://github.com/PolynomialDivision/babeld-utils/blob/master/src/main.c#L366

Sadly, this is not the easiest way in babeld since of their distance vector architecture.

I also have a shell version, but I thought if we have mor than 1200 routes, the processing with an extra tool is better.

spolack, 2021-03-06:
I dont understand who that could help on this matter?

basically its just a babel config thing.

PolynomialDivision, 2021-03-07:
> I dont understand who that could help on this matter?

Does not help, that is why I wrote `btw`. But accessing gateways in babeld seems a quite interesring thing. ;)