freifunk-berlin/bbb-configs #22: Implement Layer 3 Roaming

PolynomialDivision, 2021-04-09:
# TBD

spolack, 2021-04-21:
Conditions for easy Roaming:
* DHCP Leasetime < ARP/NDP cache lifetime
* mac address of default gateway is equal across networks to roam in

If both conditions are given it should be quite easy to listen for packets on the Router to roam to for packets which source IP isnt inside current network. If its seing such a packet, only a /128, /32 route has to be configured and advertised, in order to make roaming work until the next dhcp lease is requested.

spolack, 2021-04-21:
That will fix 95% of the common roaming scenarios. It might race if you roam to fast forth and back between networks or roam over two networks in a row, as the withdrawal shoudnt be too tight.

spolack, 2021-08-12:
Gehört das in dieses Repo? Bitte schließen falls nicht

closed 2021-08-12 by PolynomialDivision