freifunk-berlin/bbb-configs #26: add script neight.sh to the images

pmelange, 2021-06-09:
Without the script, neigh.sh, it is not very easy to debug problems on a node.  The bbb-configs router do not have this automatically installed.  It would be highly useful to have this script.

spolack, 2021-06-14:
I agree with @Akira25, this shall be rather packaged.

PolynomialDivision, 2021-06-14:
> I agree with @Akira25, this shall be rather packaged.

Please feel free to add this as PR to openwrt/routing.

Akira25, 2021-06-14:
`neigh.sh` is packages in `falter-common-olsr`.
https://github.com/Freifunk-Spalter/packages/tree/master/packages/falter-common/files-olsrv1/usr/bin

It should be enough to install that package in bbb-config-images too, shouldn't it?

spolack, 2021-07-29:
Is fixed with #44

Akira25, 2021-07-29:
please add the alias `neigh.sh` from falter-common too. Most users know this script at the old name and signalised, that they'd like to keep it that way.

spolack, 2021-07-29:
That symlink should be provided from the upstream package isnt it?

Akira25, 2021-07-29:
no it shouldn't. I think its berlin-specific. In addition I don't like to wait again 3 weeks for just copy-and-paste to be merged. This alias is defined in the package `falter-common` installing it on the images should just be fine.

spolack, 2021-07-29:
As falter-common will contains that alias, i'd propose to pull that in.

You can orientate on the following change.

https://github.com/Freifunk-Spalter/bbb-configs/commit/b6930b3d9bc2695f001013f3e287cbe4ba1c334b

spolack, 2021-07-29:
@Akira25 provided a PR #50 which has been merged.  From now on new images will contain that alias.

closed 2021-07-29 by spolack