freifunk-berlin/bbb-configs #32: load-balancing OWM-updates

SvenRoederer, 2021-06-12:
By following d115c261b8e13625b09deefdb7a7788a89669c3b I just saw the hardcoded time when OWM-updates are pushed in  https://github.com/Freifunk-Spalter/bbb-configs/blob/1346e70c88d1cf97ed3abf38e4787e3625ec8caa/roles/cfg_openwrt/files/corerouter/crontabs/root#L1

The initial intention was to have every node using a random time to update its data, so some kind of even distribution should be archived. Avoiding a load-peak as 100 nodes update at exact the same moment.

Akira25, 2021-06-14:
@spolack @nick Is it really necessary to hardcode that stuff in the bbb-configs? The owm-package will handle the crontab-entry by itself upon installation. It will set a random minute of the hour as push-time then.

spolack, 2021-06-14:
I didnt notice there was some randomization implement orginally. Thank you for the hint. Are we already hitting criticial limits?

I created that file, in order to quickly bring up airmax_dfs_reset, of which a proper packaging is outstanding.

Until this is done i'd keep that as it is, since openwrt doesnt provide a ability to put you cronjabs into different files. (cron.d)

spolack, 2021-06-14:
If the load is rather critical, we could template that crontab and randomize at rendering time in the mean time.

Akira25, 2021-06-14:
Maybe I didn't get the problem right, but couldn't you just install owm-script after you had installed airmax_dfs_reset? It adds itself to the crontab:
https://github.com/Freifunk-Spalter/packages/blob/master/luci/luci-app-falter-owm/files/owm-defaults

spolack, 2021-06-14:
The issue is, that the files are being injected *after* the packages have been installed. If we simply remove that crontab entry, we would end up without that.

SvenRoederer, 2021-06-14:
In https://github.com/openwrt/packages/pull/15667 & https://github.com/openwrt/packages/pull/13345 there is some discussion on "cron.d"-style crontab-file (as you know from most fat-unix systems). This feature can solve your loop, by just maintaining a "owm crontab" which can have a package-default, that can safely manipulated later by other scripts / tools too.

commit 45b3587c (2021-09-29): corerouter: stop shipping crontab. fixes #32
closed 2021-09-29 by spolack