freifunk-berlin/bbb-configs #42: add dns record for each antenna

booo, 2021-07-08:
We should add a dns record for each antenna so we don't have to lookup the IPs every time we want to access the device. Maybe we should announce the records to the mesh too.

spolack, 2021-07-13:
Could you please provide some sample config?

pmelange, 2021-07-24:
```
config LoadPlugin
	option library 'olsrd_nameservice'
	option suffix '.olsr'
	option hosts_file '/tmp/hosts/olsr'
	option latlon_file '/var/run/latlon.js'
	option services_file '/var/etc/services.olsr'
	list hosts '10.31.66.2 segen-switch'
	list hosts '10.31.66.11 segen-s2'
	list hosts '10.31.66.12 segen-w2'
	list hosts '10.31.66.13 segen-n2'
	list hosts '10.31.66.14 segen-o2'
	list hosts '10.31.6.18 segen-saarbr'
	list hosts '10.31.6.22 segen-f2a'
	list hosts '10.31.6.26 segen-mabb'
	list hosts '10.31.6.30 segen-sw'
	list hosts '10.31.6.34 segen-beuth'
	list hosts '10.31.6.38 segen-k12'
	list hosts '10.31.6.42 segen-no'
	list hosts '10.31.6.46 segen-wsee'
	list hosts '10.31.6.50 segen-s5'
	list hosts '10.31.6.54 segen-w5'
	list hosts '10.31.6.58 segen-n5'
	list hosts '10.31.6.62 segen-o5'
```

spolack, 2021-07-25:
Ok, the diff with the change i've prepared looks similar:
```
--- before: /tmp/ansible-openwrt/configs/fesev-core/etc//config/olsrd
+++ after: /home/spolack/.ansible/tmp/ansible-local-103810n4kvb5l/tmpwgg952hh/olsrd.j2
@@ -7,6 +7,11 @@
 	option hosts_file '/tmp/hosts/olsr'
 	option latlon_file '/var/run/latlon.js'
 	option services_file '/var/etc/services.olsr'
+	list hosts '10.31.96.242 fesev-switch'
+	list hosts '10.31.96.243 fesev-sama'
+	list hosts '10.31.96.244 fesev-ap1'
+	list hosts '10.31.96.245 fesev-ap2'
+	list hosts '10.31.96.246 fesev-ap3'
 
 config LoadPlugin
 	option accept '0.0.0.0'

changed: [fesev-core] => (item=/home/spolack/git/bbb-configs/roles/cfg_openwrt/templates/corerouter/config/olsrd.j2)
--- before: /tmp/ansible-openwrt/configs/emma-core/etc//config/olsrd
+++ after: /home/spolack/.ansible/tmp/ansible-local-103810n4kvb5l/tmp2581xo3x/olsrd.j2
@@ -7,6 +7,10 @@
 	option hosts_file '/tmp/hosts/olsr'
 	option latlon_file '/var/run/latlon.js'
 	option services_file '/var/etc/services.olsr'
+	list hosts '10.31.11.10 emma-switch01'
+	list hosts '10.31.11.11 emma-switch02'
+	list hosts '10.31.11.12 emma-switch03'
+	list hosts '10.31.11.13 emma-switch04'
```

spolack, 2021-07-25:
Tested on sama. works!

commit 8e13400e (2021-07-25): corerouter: add dns record for every assignment

fixes #42
closed 2021-07-25 by spolack