freifunk-berlin/bbb-configs #43: add a LICENSE file

booo, 2021-07-08:
Is the license of this project mentioned somewhere?

Anyway we should add a LICENSE file to the root of the repository.

PolynomialDivision, 2021-07-08:
I'm interested why you see a need for a lincense file. Of course we can do it. :)

booo, 2021-07-08:
Because closed source software is bullshit and we should not encourage people to write closed source software or software without proper license for further use. For a more precise definition of bullshit please use your favourite.

PolynomialDivision, 2021-07-08:
I would consider @spolack is the right person who should add license file here. I really don't care.

spolack, 2021-07-28:
Any proposals? GPLv3?

PolynomialDivision, 2021-07-28:
> Any proposals? GPLv3?

Depends on what you want to achieve with it. We have to be carefully not putting the actual ips and configs, but only the ansible configuration into GPLv3. Otherwise people maybe have trouble using this opensource project for their private homes or companies.

spolack, 2021-08-04:
Ǹot sure whether its possible to put IPs under GPL. I'm not familiar with that topic. @booo Something you'd suggest?

kls0e, 2021-09-15:
in this case, [WTFPL ](https://de.wikipedia.org/wiki/WTFPL#Verwendung_der_Lizenz) might be an option.

Akira25, 2021-09-19:
> > Any proposals? GPLv3?
> 
> Depends on what you want to achieve with it. We have to be carefully not putting the actual ips and configs, but only the ansible generation into GPLv3. Otherwise people maybe have trouble using this opensource project for their private homes or companies.

According to this [article](https://de.wikipedia.org/wiki/GNU_General_Public_License#Nutzung) in the German Wikipedia, its only mandatory to publish the (modified) source code, if you plan to sell or distribute the software. Private (internal) use is possible every time.

Any modification anyone likes is possible with GPLv3. The only obligation is to publish your modifications under the same license, to keep the software free. If you want to, you are free to sell the software too, but you must publish it then.

Concerning the IP-Addresses: I doubt that an IP-Address reach ["Schöpfungshöhe"](https://de.wikipedia.org/wiki/Sch%C3%B6pfungsh%C3%B6he). Thus they cannot be licensed.

PR #62 (2021-09-24): Create LICENSE.md
Add a LICENSE for the ansible-repo. 

Fixes #43
closed 2021-09-24 by pmelange