freifunk-berlin/bbb-configs #49: Add hostname support for even more devices

pmelange, 2021-07-26:
For example, at hds, there could be a hostname hds-switch
Or even better, at emma there could be hostnames hds-switch0 hds-switch1 hds-switch2...

PolynomialDivision, 2021-07-26:
As I read the ansible script, this is announced? Just to the assignment in the network config?

pmelange, 2021-07-26:
These are network devices, not mesh devices.  So no, it doesn't go in /etc/config/network.  Just in the olsrd nameservice config

PolynomialDivision, 2021-07-26:
https://github.com/Freifunk-Spalter/bbb-configs/commit/8e13400e4fb35ffe073f40bf3b624bcc032a7d5a

It does not filter on mesh?

pmelange, 2021-07-26:
I don't understand your question

spolack, 2021-07-28:
No one hinders you on describing them in assignments. Nick pointed you to the template, where the vars are being rendered.

Not an issue tbh

spolack, 2021-09-20:
Closing as this feature is implemented. It only needs to be used.

closed 2021-09-20 by spolack