freifunk-berlin/bbb-configs #52: Feature: new role mesh-ap

Akira25, 2021-08-02:
In some setips, we want to broadcast mesh and client wifi on the same AP, similar to the setup in standard falter. It would be cool to have a role in bbb-configs enabling this configuration.

spolack, 2021-08-02:
This is already possible and is being used on kiehlufer and scharni. The 11s is being bridged to the core router, which takes care of routing.

https://github.com/Freifunk-Spalter/bbb-configs/blob/master/group_vars/location_kiehlufer/networks.yml

```
- vid: 20
    role: mesh
    name: mesh_kiehlufer
    prefix: 10.31.82.142/32
    ipv6_subprefix: -4
    mesh_metric: 2048
    mesh_ap: kiehlufer-nf-wbp2
    mesh_radio: 11a_standard
    mesh_iface: mesh

```

Its not yet tested, whether it would work what would happen if the core router sees itself over mesh link. e.g. mesh interface of ap1 links via 11s to ap2, which operates at same channel.  I think even if it would break that would be a solvable problem.

Akira25, 2021-09-01:
then we can close this.

closed 2021-09-01 by Akira25

PolynomialDivision, 2021-09-05:
> This is already possible for role_ap (and perhaps role_corerouter (untested)) and is being used on kiehlufer and scharni. The 11s mesh is bridged to the core router, which takes care of routing.
> 
> https://github.com/Freifunk-Spalter/bbb-configs/blob/master/group_vars/location_kiehlufer/networks.yml
> 
> ```
> - vid: 20
>     role: mesh
>     name: mesh_kiehlufer
>     prefix: 10.31.82.142/32
>     ipv6_subprefix: -4
>     mesh_metric: 2048
>     mesh_ap: kiehlufer-nf-wbp2
>     mesh_radio: 11a_standard
>     mesh_iface: mesh
> ```
> 
> Its not yet tested, whether it would work what would happen if the core router sees itself over mesh link. e.g. mesh interface of ap1 links via 11s to ap2, which operates at same channel. I think even if it would break that would be a solvable problem.

I will have a look at this.