freifunk-berlin/bbb-configs #54: uplink-gw: Default Route is not withdrawn from olsr in case of broken uplink

spolack, 2021-08-24:
```
root@saarbruecker-gw:~# ping 80.67.169.40
PING 80.67.169.40 (80.67.169.40): 56 data bytes
^C
--- 80.67.169.40 ping statistics ---
1 packets transmitted, 0 packets received, 100% packet loss
root@saarbruecker-gw:~# ping 80.67.169.40
PING 80.67.169.40 (80.67.169.40): 56 data bytes
```

```
config LoadPlugin
	option library 'olsrd_dyn_gw'
	list Ping '46.182.19.48'
	list Ping '80.67.169.40'
	list Ping '194.150.168.168'
	option ignore '0'
	option PingCmd 'ping -c 1 -q -I lan3 %s'
	option PingInterval '30'
```

```
Tue Aug 24 21:31:48 2021 daemon.info olsrd: /etc/init.d/olsrd: olsrd_setup_smartgw_rules() Notice: Inserting firewall rules for SmartGateway
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '1' (was 1) to /proc/sys/net/ipv4/ip_forward
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tunl0/rp_filter
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/all/send_redirects
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/all/rp_filter
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/lan0.10/send_redirects
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan0.10/rp_filter
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Adding interface lan0.10
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: New main address: 10.31.83.56
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/lan0.11/send_redirects
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan0.11/rp_filter
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Adding interface lan0.11
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/lan0.12/send_redirects
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/lan0.12/rp_filter
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: Adding interface lan0.12
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: olsr.org - pre-0.9.9-git_0000000-hash_b5353b42d6a718a7c674facbe2cb5536 successfully started
Tue Aug 24 21:31:48 2021 daemon.info olsrd[9616]: You might have another node with main ip 10.31.83.56 in the mesh!
Tue Aug 24 21:32:04 2021 daemon.err collectd[2717]: olsrd plugin: connect failed: Connection refused
Tue Aug 24 21:32:04 2021 daemon.err collectd[2717]: read-function of plugin `olsrd' failed. Will suspend it for 60.000 seconds.
Tue Aug 24 21:32:24 2021 daemon.info olsrd[9616]: Tunnel tnl_0a1f5539 added, to 10.31.85.57
Tue Aug 24 21:33:04 2021 daemon.err collectd[2717]: olsrd plugin: connect failed: Connection refused
Tue Aug 24 21:33:04 2021 daemon.err collectd[2717]: read-function of plugin `olsrd' failed. Will suspend it for 120.000 seconds.
```

spolack, 2021-09-14:
@pmelange do you have any idea, why olsrd is still advertising the default route when the targets stop being reachable?

pmelange, 2021-09-14:
I don't see any problems with the config.  The line "You might have another node with main ip 10.31.83.56 in the mesh!" worries me though.  That IP and 10.31.83.57 both route to k12-ff-4.olsr and k12-ff-praxis.olsr

I don't have this problem with the 1.1.2 and 1.2.0-snapshot falter firmware.

pmelange, 2021-09-14:
I think k12 made a typo. saarbr has the right addresses (and possible extra addresses that we are not useing)
```
pmelange@config:~$ nipap address list "saarbr"
Searching for prefixes in any VRF...
VRF RT  Prefix                                     #   Node  Order ID  Customer ID                 Description                             
===========================================================================================================================================
-       10.31.0.0/16                            R  -   None  None      None                        Mesh IPs                                
-         10.31.83.48/29                        A  #1  None  None      trashaddy94@googlemail.com  saarbruecker-mgmt                       
-         10.31.83.56/30                        A  #1  None  None      trashaddy94@googlemail.com  saarbruecker-mesh                       
-       10.230.0.0/16                           R  -   None  None      None                        HNA                                     
-         10.230.230.64/26                      A  #1  None  None      t.voss@syseleven.de         saarbrueckerstr                         
-       2001:bf7::/32                           R  -   None  None      None                        all prefixes                            
-         2001:bf7:760::/44                     R  -   None  None      Berlin                      Freifunk Berlin (Wahlkreis 76, Pankow)  
-           2001:bf7:760:2200::/56              A  #1  None  None      trashaddy94@googlemail.com  saarbruecker-v6

spolack, 2021-09-14:
Please revoke that prefix if possible. 10.230.230.64/26

Theo doesnt need that prefix for saarbruecker anymore :)

pmelange, 2021-09-14:
I don't know how.  Could you ask Theo to do it?

pmelange, 2021-09-14:
I just restart olsrd on saarbr and I still get the following:

```Tue Sep 14 15:12:59 2021 daemon.info olsrd[27430]: You might have another node with main ip 10.31.83.56 in the mesh!```

If there is another node, with the same IP address and advertising a gw, then the cause of the problem is probably that

spolack, 2021-09-14:
> I don't know how. Could you ask Theo to do it?

-->  Die IP-Adressen für saarbrueckerstr wurden soeben gelöscht.  :-)

spolack, 2021-12-24:
Last time @pmelange and me checked it was working.

closed 2021-12-24 by spolack