freifunk-berlin/bbb-configs #60: hds-core shows a 404 error for http://hds-core.olsr/cgi-bin/luci/freifunk/olsr/neighbors

pmelange, 2021-09-10:
The olsr status pages are not available on hds-core.

spolack, 2021-09-10:
@freifunkoperator  Please fix. Please dont remove essential packages on device level!

https://github.com/Freifunk-Spalter/bbb-configs/blob/cff3a5985380da1e16412eeaab7a6d770ac60ae6/group_vars/model_mikrotik_routerboard_750gr3.yml#L7

spolack, 2021-09-14:
@freifunkoperator you can also mention the issue number on the commit message to create references.

Please also roll out your change in order to fix that issue.

https://github.com/Freifunk-Spalter/bbb-configs/commit/9ce200538534717230d6e04b1c138faa3fe3ca9d

Akira25, 2021-09-19:
AFAICT the changes were rolled out to hds-core. The router now says, its falter instead of OpenWrt. The commit hash is the most recent one. So we can close this issue.

[image]

closed 2021-09-19 by Akira25