freifunk-berlin/bbb-configs #66: meshing on core router doesnt require setting up bridge and vlan

spolack, 2021-09-25:
That could be simplified
```/etc/config/wireless
config wifi-iface 'radio0_if2'
	option device 'radio0'
	option ifname 'wlan5-mesh'
	option mode 'mesh'
	option mesh_id 'Mesh-Freifunk-Berlin'
	option mesh_fwding '0'
	option mcast_rate '12000'
	option network 'mesh_core'
```

```/etc/config/network
config interface 'mesh_core'
	option ifname 'eth0.20'
	option proto 'static'
	option ipaddr '10.31.114.0/32'
        option type 'bridge'
	option ip6addr '2001:bf7:830:26ff::1/128'
```

pmelange, 2021-09-26:
First of all, the following are no longer valid:
```
config interface 'mesh_core'
     option ifname 'eth0.20' # ifname is not longer a valid option for an interface
     option type 'bridge' # type is no longer a valid option for an interface
```

but having the following works with my tests and is the way the standard firmware has always been working:
```
config wifi-iface 'wlan0_mesh_2'
	option device 'radio0'
	option ifname 'wlan0-mesh-2'
	option network 'wireless0'
	option encryption 'none'
	option mcast_rate '12000'
	option mesh_fwding '0'
	option mode 'mesh'
	option mesh_id 'Mesh-Freifunk-Berlin'
```
```
config interface 'wireless0'
	option netmask '255.255.255.255'
	option proto 'static'
	option ipaddr '10.31.34.68'
	option ip6assign '64'
```

spolack, 2021-12-13:
Its implemented. Result is visible in huette-core config

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option dns '46.182.19.48 80.67.169.40 194.150.168.168 2001:4ce8::53 2001:910:800::12'
	option ip6addr '2001:bf7:830:2600::/56'

config interface 'mesh_core'
	option proto 'static'
	option ipaddr '10.31.114.0/32'
	option ip6addr '2001:bf7:830:26ff::1/128'

config interface 'mesh_huette'
	option device 'eth0.21'
	option proto 'static'
	option ipaddr '10.31.114.1/32'
	option ip6addr '2001:bf7:830:26fe::1/128'

config interface 'dhcp'
	option device 'br-dhcp'
	option proto 'static'
	option ipaddr '10.31.113.129/25'
	option macaddr '02:00:00:00:00:01'
	option ip6addr '2001:bf7:830:2600::1/64'

config device
	option name 'br-dhcp'
        option type 'bridge'
	list ports 'eth0.40'

config interface 'mgmt'
	option device 'eth0.42'
	option proto 'static'
	option ipaddr '10.31.112.113/28'
	option ip6addr '2001:bf7:830:2601::1/64'
```

spolack, 2021-12-13:
Fixed by 0ffabd9

closed 2021-12-13 by spolack