freifunk-berlin/bbb-configs #75: ipq40xx: Consider tweaking cpufreq scaling in order to optimize stability and performance

spolack, 2021-09-29:
Some OpenWRT forum posts mention that increasing the cpu min_freq increases the performance in terms of throughput and jitter significatly.

In addition i've found the sama to be unstable and kept rebooting every couple of minutes while i stream ARD Mediathek LiveTV from wilgu10 over sama. Once i've set the govonor to performance it was stable.

On zyxel nbg6617 the stockfirmware configures the following settings as below according to https://forum.openwrt.org/t/ipq40xx-ea6350-v3-as-wap-wifi-to-lan-latency/51623/11

```
	# change scaling governor as ondemand to enable clock scaling based on system load
	echo "ondemand" > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor

	# set scaling min freq as 200 MHz
	echo "200000" > /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq

	# Change sampling rate for frequency scaling decisions to 1s, from 10 ms
	echo "1000000" > /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate

	# Change sampling rate for frequency down scaling decision to 10s
	echo 10 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_down_factor

	# Change the CPU load threshold above which frequency is up-scaled to
	# turbo frequency,to 50%
	echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
```

PolynomialDivision, 2021-09-29:
I'm fine with that.

robertfoss, 2021-09-29:
Later in the thread someone tested[1] using the `performance` governor, which seemed to improve local ping latency from ~2.2ms to ~0.5ms.

I think a usecase like ping might allow the cores time to spin down to a lower frequency when using the 'ondemand' governor, due the amount of time before each individual ping. If the testing was done during modest load instead of the essentially idle load pattern of ping, I think there would be less difference between `ondemand` and `performance`.

[1] https://forum.openwrt.org/t/ipq40xx-ea6350-v3-as-wap-wifi-to-lan-latency/51623/37

PolynomialDivision, 2021-09-29:
Any suggestions how we can systematically test it, allowing us to go for a well informed decision?

spolack, 2021-09-29:
> If the testing was done during modest load instead of the essentially idle load pattern of ping, I think there would be less difference between `ondemand` and `performance`.

I agree, but even low throughput load can be time-critical. I think of VoIP or online gaming.

PolynomialDivision, 2021-09-29:
https://github.com/Freifunk-Spalter/bbb-configs/pull/77

PolynomialDivision, 2021-09-29:
https://github.com/Freifunk-Spalter/bbb-configs/pull/78

robertfoss, 2021-09-30:
@spolack I think the `nbg6617`config seems pretty reasonable.

If hit any issues we can revert it, and if the performance isn't good enough we can always spend more time on it.

spolack, 2021-09-30:
Seems the scaling doesnt behave as configured:

```
root@sama-core:~# echo 1 > /sys/devices/system/cpu/cpufreq/policy0/stats/reset
root@sama-core:~# while true; do date; cat /sys/devices/system/cpu/cpufreq/policy0/stats/time_in_state; sleep 1; done
Thu Sep 30 20:52:20 CEST 2021
48000 0
200000 1926
500000 200
716000 282
Thu Sep 30 20:52:21 CEST 2021
48000 0
200000 2030
500000 200
716000 282
Thu Sep 30 20:52:22 CEST 2021
48000 0
200000 2134
500000 200
716000 282
Thu Sep 30 20:52:23 CEST 2021
48000 0
200000 2239
500000 200
716000 282
Thu Sep 30 20:52:24 CEST 2021
48000 0
200000 2343
500000 200
716000 282
Thu Sep 30 20:52:25 CEST 2021
48000 0
200000 2447
500000 200
716000 282
Thu Sep 30 20:52:26 CEST 2021
48000 0
200000 2526
500000 200
716000 305
Thu Sep 30 20:52:27 CEST 2021
48000 0
200000 2526
500000 200
716000 406
Thu Sep 30 20:52:28 CEST 2021
48000 0
200000 2612
500000 200
716000 423
Thu Sep 30 20:52:29 CEST 2021
48000 0
200000 2627
500000 200
716000 511
Thu Sep 30 20:52:30 CEST 2021
48000 0
200000 2676
500000 200
716000 565
Thu Sep 30 20:52:31 CEST 2021
48000 0
200000 2727
500000 200
716000 616
Thu Sep 30 20:52:32 CEST 2021
48000 0
200000 2740
500000 200
716000 707
Thu Sep 30 20:52:33 CEST 2021
48000 0
200000 2827
500000 200
716000 722
Thu Sep 30 20:52:34 CEST 2021
48000 0
200000 2827
500000 200
716000 823
Thu Sep 30 20:52:35 CEST 2021
48000 0
200000 2905
500000 200
716000 848
Thu Sep 30 20:52:36 CEST 2021
48000 0
200000 2927
500000 200
716000 928
```

On  20:52:26 i started an iperf3 through sama. You can see the cpufreq was ranning on 716MHz.  According to the config it should wait 10s before scaling down. In reality it already scaled down after 2 seconds.

robertfoss, 2021-09-30:
@spolack When in the above log does the ip3rf3 run end? Or is that not a part of the log?

spolack, 2021-09-30:
I stopped the loop right after iperf3 has ended..

robertfoss, 2021-09-30:
Alright, but apart from scaling back down too soon it seems to work.

48Mhz is never used
200Mhz is the lowest rate used
716MHz is used when iperf3 induces a higher load

spolack, 2021-09-30:
I even tried to increase the sampling rate (and would then expect everything to scale with even more delay, but the behavior doesnt feel like it has changed)

spolack, 2021-09-30:
I also tested governor 'performance' which works as designed.

robertfoss, 2021-10-01:
This is can be closed since #78 was merged.

closed 2021-10-01 by PolynomialDivision