freifunk-berlin/bbb-configs #76: inventory: construct group assignments based on vars which are inherited from through dynamic group assignments

spolack, 2021-09-29:
Currently its only possible to build group assignments from host_vars, which are in the scope of the host. This requires us to specifcy all those relations inside the host_vars. E.g.:

Example of the real relaition of the variables: 
```
model of sama-core is fritzbox-4440
target of fritzbox-4440 is ipq40xx_generic
```

But instead of defining the target inside `group_vars/model_*`, it has to be defined in the host_vars of every individual host.

We could avoid that by running the constructed plugin multiple times.
* First run inherits all variables from the model group assignment (variables are merged and flattend during that process)
* In addition a second run would inherit all variables from target group, which assignment is now built based from the target variable inside the model

Implications of that approach:
* As the target construction is running as the very last time it could break the ansible variable precedence, if variables are overriden from within the target_ group_vars.
* Multiple passes of the inventory plugin will likely needs more time to process the inventory.

spolack, 2021-09-29:
Quick Benchmark before:
```
spolack@spolack:~/git/bbb-configs$ time ansible-inventory --graph --vars > /dev/null

real	0m14,717s
user	0m11,684s
sys	0m6,024s
spolack@spolack:~/git/bbb-configs$ time ansible-inventory --graph --vars > /dev/null

real	0m12,983s
user	0m11,109s
sys	0m4,509s

spolack@spolack:~/git/bbb-configs$ time ansible-inventory --graph --vars > /dev/null

real	0m12,025s
user	0m10,011s
sys	0m4,472s
```

spolack, 2021-09-29:
After the change its even faster. Dont really know, but it seems the change wont drastically decrease the performance.

closed 2021-09-29 by spolack

PolynomialDivision, 2021-09-29:
> After the change its even faster. Dont really know, but it seems the change wont drastically decrease the performance.

Did you deleted any cache before testing?