freifunk-berlin/bbb-configs #96: uplink_gateways: conntrackd goes bad

spolack, 2021-11-16:
I've observed for the second time that conntrackd behaves wrongly after some time.

```
root@ohlauer-gw:~# conntrackd -s
cache internal:
current active connections:	         773
connections created:		        1181	failed:	           0
connections updated:		        1205	failed:	           0
connections destroyed:		         408	failed:	           0

external inject:
connections created:		        1244	failed:	           0
connections updated:		        2015	failed:	           0
connections destroyed:		          14	failed:	           0

traffic processed:
                   0 Bytes                         0 Pckts

UDP traffic (active device=lo):
             1113212 Bytes sent               347292 Bytes recv
                1897 Pckts sent                 3696 Pckts recv
                   0 Error send                    0 Error recv

message tracking:
                   3 Malformed msgs                  116 Lost msgs
```

In case it is broken you can see that no conntrack entries are injected externally.

It can be fixed be restarting conntrackd on the other node(s)

```
root@saarbruecker-gw:~# /etc/init.d/conntrackd restart`
``

We should either monitor this or implement a workaround until monitoring is in place.

PolynomialDivision, 2021-11-16:
How does a workaround looks like?

spolack, 2021-11-16:
Restart it unconditionally all x hours?

PolynomialDivision, 2021-11-17:
Bugtracker:
https://bugzilla.netfilter.org/buglist.cgi?quicksearch=conntrackd

You can look thorugh the history and see that OpenWrt already importaed some bugfix, however I don't understand the impact:
- https://git.netfilter.org/conntrack-tools/commit/?id=b55717d46ae3b7c3769192a66e565bc7c2d833a1

spolack, 2021-11-17:
It broke again:
```
root@saarbruecker-gw:~# conntrackd -s
cache internal:
current active connections:	          14
connections created:		      149531	failed:	     2418387
connections updated:		     1039873	failed:	           0
connections destroyed:		      149517	failed:	           0

external inject:
connections created:		        8173	failed:	           6
connections updated:		       16157	failed:	           0
connections destroyed:		         463	failed:	           0

traffic processed:
                   0 Bytes                         0 Pckts

UDP traffic (active device=lo):
           320180076 Bytes sent              2587632 Bytes recv
              438460 Pckts sent                29517 Pckts recv
                   0 Error send                    0 Error recv

message tracking:
                  35 Malformed msgs                 9850 Lost msgs
```

Only the connections created: failed counter is rising.

spolack, 2021-11-17:
> You can look through the history and see that OpenWrt already imported some bugfix, however I don't understand the impact:
I updated conntrack on both gateways, please have an eye on it:

```
root@ohlauer-gw:/tmp# wget https://downloads.openwrt.org/snapshots/packages/mips64_octeonplu
s/packages/conntrack_1.4.6-2_mips64_octeonplus.ipk
Downloading 'https://downloads.openwrt.org/snapshots/packages/mips64_octeonplus/packages/conntrack_1.4.6-2_mips64_octeonplus.ipk'
Connecting to 168.119.138.211:443
Writing to 'conntrack_1.4.6-2_mips64_octeonplus.ipk'
conntrack_1.4.6-2_mi 100% |*******************************| 21944   0:00:00 ETA
Download completed (21944 bytes)
root@ohlauer-gw:/tmp# wget https://downloads.openwrt.org/snapshots/packages/mips64_octeonplu
s/packages/conntrackd_1.4.6-2_mips64_octeonplus.ipk
Downloading 'https://downloads.openwrt.org/snapshots/packages/mips64_octeonplus/packages/conntrackd_1.4.6-2_mips64_octeonplus.ipk'
Connecting to 168.119.138.211:443
Writing to 'conntrackd_1.4.6-2_mips64_octeonplus.ipk'
conntrackd_1.4.6-2_m 100% |*******************************| 77938   0:00:00 ETA
Download completed (77938 bytes)
root@ohlauer-gw:/tmp# opkg install *.ipk
Installing conntrack (1.4.6-2) to root...
Upgrading conntrackd on root from 1.4.6-1 to 1.4.6-2...
Configuring conntrackd.
Configuring conntrack.
root@ohlauer-gw:/tmp# reboot
root@ohlauer-gw:/tmp# Connection to 176.74.57.19 closed by remote host.
Connection to 176.74.57.19 closed.
```

spolack, 2021-11-17:
It might be possible that Mode FTFW is not the best choice for one to many replication.

```
This mode is based on a reliable protocol that performs message tracking. Thus, the protocol can recover from message loss, re-ordering and corruption.
```

We should rather migrate to Mode NOTRACK

```
Mode NOTRACK

Is the most simple mode as it is based on a best effort replication protocol, ie. unreliable protocol. This protocol sends and receives the state information without performing any specific checking.

In this synchronization mode you may configure DisableInternalCache, DisableExternalCache, CommitTimeout, PurgeTimeout and StartupResync.

DisableInternalCache <on|off>
    This clause allows you to disable the internal cache. Thus, the synchronization messages are directly sent through the dedicated link.

    This option is set off by default.

DisableExternalCache <on|off>
    Same as in FTFW mode.

CommitTimeout <seconds>
    Same as in FTFW mode.

PurgeTimeout <seconds>
    Same as in FTFW mode.

StartupResync <on|off>
    Same as in FTFW mode.
```

spolack, 2021-11-17:
https://www.spinics.net/lists/netfilter/msg60409.html
...

PolynomialDivision, 2021-11-18:
Why almost-fix? I made a WIP commit:

https://github.com/openwrt/packages/pull/17148

PolynomialDivision, 2021-11-18:
Updated both gateways.

PR #97 (2021-11-21): uplink_gateway: switch to NOTRACK
FTFW:
This mode is based on a reliable protocol that performs message
tracking. Thus, the protocol can recover from message loss, re-ordering
and corruption.

NOTRACK
Is the most simple mode as it is based on a best effort replication
protocol, ie. unreliable protocol. This protocol sends and receives
the state information without performing any specific checking.

Source: https://manpages.debian.org/testing/conntrackd/conntrackd.conf.5.en.html

We use samplicator with UDP so we should switch to NOTRACK.

Fixes: #96 (uplink_gateways: conntrackd goes bad)
commit dbd1c83c (2021-11-21): uplink_gateway: switch to NOTRACK

FTFW:
This mode is based on a reliable protocol that performs message
tracking. Thus, the protocol can recover from message loss, re-ordering
and corruption.

NOTRACK
Is the most simple mode as it is based on a best effort replication
protocol, ie. unreliable protocol. This protocol sends and receives
the state information without performing any specific checking.

Source: https://manpages.debian.org/testing/conntrackd/conntrackd.conf.5.en.html

We use samplicator with UDP so we should switch to NOTRACK.

Fixes: #96 (uplink_gateways: conntrackd goes bad)
closed 2021-11-21 by PolynomialDivision