freifunk-berlin/bbb-configs #100: /tmp/hosts is not being created, causing olsrd based hostnames not to work.

pmelange, 2021-11-22:


PolynomialDivision, 2021-11-22:
I don't know if there is a reson why @spolack disalbed dnsmasq? I think even this device will not function as dns server without it?

pmelange, 2021-11-22:
If it's a worry about having a public facing dns server, you can use option ```notinterface``` on the public interfaces.

I find it useful to be able to do ```iperf3 -c hostname.olsr`` and not have to know the IP.

spolack, 2021-12-24:
I didnt work on that, but it seems to be fixed. 

```
root@saarbruecker-gw:~# ps w | grep dns
 1620 dnsmasq   1712 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
```

```
root@saarbruecker-gw:~# ping sama-core.olsr
PING sama-core.olsr (10.31.79.176): 56 data bytes
64 bytes from 10.31.79.176: seq=0 ttl=64 time=1.047 ms
64 bytes from 10.31.79.176: seq=1 ttl=64 time=0.922 ms
^C
```

closed 2021-12-24 by spolack