freifunk-berlin/bbb-configs #111: ath9k stacktraces /net/mac80211/rate.c:677

spolack, 2021-12-07:
On magda-ap4 (nsm2) we regulary see stacktraces, altough it doesnt break the wifi it should be investigated.

```
root@magda-ap4:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02-SNAPSHOT'
DISTRIB_REVISION='r16335-4b0f87729c'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02-SNAPSHOT r16335-4b0f87729c'
DISTRIB_TAINTS=''

root@magda-ap4:~# uname -a
Linux magda-ap4 5.4.158 #0 Sun Nov 7 18:21:40 2021 mips GNU/Linux
```

```
[1047176.077032] WARNING: CPU: 0 PID: 0 at backports-5.10.68-1/net/mac80211/rate.c:677 0x8161e700 [mac80211@c820b7db+0x7cdb0]
[1047176.088134] Modules linked in: ath9k ath9k_common ath9k_hw ath nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_ct nft_counter nf_tables_set nf_tables nf_nat nf_conntrack mac80211 lzo cfg80211 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 lzo_decompress lzo_compress compat zram zsmalloc crypto_acompress gpio_button_hotplug
[1047176.129369] CPU: 0 PID: 0 Comm: swapper Tainted: G        W         5.4.158 #0
[1047176.136790] Stack : 81680000 800b98ec 80660000 805fa4b8 00000000 00000000 00000000 00000000
[1047176.145372]         00000000 00000000 00000000 00000000 00000000 00000001 81c0bc30 20cca586
[1047176.153947]         81c0bcc8 00000000 00000000 000053a4 00000038 8057a804 20202057 20202020
[1047176.162523]         000053a4 8cbaef7e 00000000 0001f959 00000000 81c0bc10 00000000 8161e700
[1047176.171098]         00000009 81676760 815cc1a0 80630000 00000003 8032022c 00000000 80790000
[1047176.179674]         ...
[1047176.182309] Call Trace:
[1047176.182331] [<81680000>] 0x81680000 [ath9k_hw@a0ef92ae+0x55560]
[1047176.191082] [<800b98ec>] 0x800b98ec
[1047176.194771] [<80660000>] 0x80660000
[1047176.198462] [<8057a804>] 0x8057a804
[1047176.202159] [<8161e700>] 0x8161e700 [mac80211@c820b7db+0x7cdb0]
[1047176.208289] [<8032022c>] 0x8032022c
[1047176.211983] [<81680000>] 0x81680000 [ath9k_hw@a0ef92ae+0x55560]
[1047176.218106] [<80069364>] 0x80069364
[1047176.221795] [<8006936c>] 0x8006936c
[1047176.225481] [<800824f4>] 0x800824f4
[1047176.229182] [<8161e700>] 0x8161e700 [mac80211@c820b7db+0x7cdb0]
[1047176.235304] [<8008259c>] 0x8008259c
[1047176.239013] [<8161e700>] 0x8161e700 [mac80211@c820b7db+0x7cdb0]
[1047176.245155] [<8162e728>] 0x8162e728 [mac80211@c820b7db+0x7cdb0]
[1047176.251279] [<800c8a7c>] 0x800c8a7c
[1047176.254968] [<800c3f10>] 0x800c3f10
[1047176.258653] [<80545ea4>] 0x80545ea4
[1047176.262351] [<816e991c>] 0x816e991c [ath9k@4226d984+0x19450]
[1047176.268230] [<816ecc84>] 0x816ecc84 [ath9k@4226d984+0x19450]
[1047176.274093] [<800a4544>] 0x800a4544
[1047176.277797] [<816e4664>] 0x816e4664 [ath9k@4226d984+0x19450]
[1047176.283680] [<81630d9c>] 0x81630d9c [mac80211@c820b7db+0x7cdb0]
[1047176.289825] [<81630ef0>] 0x81630ef0 [
This happens frequently on 
mac80211@c820b7db+0x7cdb0]
[1047176.295968] [<81638bf0>] 0x81638bf0 [mac80211@c820b7db+0x7cdb0]
[1047176.302117] [<8163a838>] 0x8163a838 [mac80211@c820b7db+0x7cdb0]
[1047176.308261] [<816e4eb8>] 0x816e4eb8 [ath9k@4226d984+0x19450]
[1047176.314142] [<80660000>] 0x80660000
[1047176.317832] [<80660000>] 0x80660000
[1047176.321518] [<80085380>] 0x80085380
[1047176.325208] [<8057fd70>] 0x8057fd70
[1047176.328893] [<803b5538>] 0x803b5538
[1047176.332574] [<802bfa2c>] 0x802bfa2c
[1047176.336263] [<80064f78>] 0x80064f78
[1047176.339943] 
[1047176.341625] ---[ end trace 6f7b6d5b39060958 ]---
```

spolack, 2021-12-09:
Also seen on WDR4900.

```
[1816890.484809] WARNING: CPU: 0 PID: 2381 at backports-5.10.68-1/net/mac80211/rate.c:677 0xc9517ea0 [mac80211@2e342341+0xa2660]
[1816890.496100] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_ct nft_counter nf_tables_set nf_tables nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc nfnetlink nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 gpio_keys leds_gpio fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd button_hotplug input_core usbcore usb_common
[1816890.582351] CPU: 0 PID: 2381 Comm: olsrd Tainted: G        W         5.4.158 #0
[1816890.589824] NIP:  c9517ea0 LR: c94a5104 CTR: c006d108
[1816890.595040] REGS: c7ff5ca8 TRAP: 0700   Tainted: G        W          (5.4.158)
[1816890.602423] MSR:  00029000 <CE,EE,ME>  CR: 48448844  XER: 00000000
[1816890.608774] 
[1816890.608774] GPR00: c94a5104 c7ff5d60 c3918000 00000100 ffffffff 00000011 00000001 00000001 
[1816890.608774] GPR08: 0000001b c4bdd628 000001e0 c4775da0 88444822 00000000 c068fcac c0690000 
[1816890.608774] GPR16: 00400100 00000000 c06b0000 c05b1b00 00000009 00000000 fffff000 00000001 
[1816890.608774] GPR24: c4ba6cf8 00000000 c4bac130 c4b1b708 c4ba6cf8 00000004 c4bdd628 c4775d80 
[1816890.644496] Call Trace:
[1816890.647112] [c7ff5d60] [c9517d4c] 0xc9517d4c [mac80211@2e342341+0xa2660] (unreliable)
[1816890.655109] [c7ff5dd0] [c94a5104] 0xc94a5104 [ath9k@b6a37847+0x1f4f8]
[1816890.661717] [c7ff5e10] [c949c980] 0xc949c980 [ath9k@b6a37847+0x1f4f8]
[1816890.668325] [c7ff5e50] [c952a8fc] 0xc952a8fc [mac80211@2e342341+0xa2660]
[1816890.675194] [c7ff5e90] [c952aae0] 0xc952aae0 [mac80211@2e342341+0xa2660]
[1816890.682063] [c7ff5eb0] [c95328e4] 0xc95328e4 [mac80211@2e342341+0xa2660]
[1816890.688932] [c7ff5f10] [c95345bc] 0xc95345bc [mac80211@2e342341+0xa2660]
[1816890.695797] [c7ff5f60] [c002e6e0] 0xc002e6e0
[1816890.700233] [c7ff5f90] [c05a97f4] 0xc05a97f4
[1816890.704668] [c7ff5ff0] [c000d294] 0xc000d294
[1816890.709102] [c4499890] [c00057b8] 0xc00057b8
[1816890.713536] [c44998b0] [c000f4ac] 0xc000f4ac
[1816890.717973] --- interrupt: 501 at 0xc004d8ac
[1816890.717973]     LR = 0xc05a5bb4
[1816890.725706] [c4499978] [c006c2c0] 0xc006c2c0 (unreliable)
[1816890.731270] [c4499998] [c05a5bb0] 0xc05a5bb0
[1816890.735705] [c44999d8] [c05a5da8] 0xc05a5da8
[1816890.740140] [c44999f8] [c05a91c8] 0xc05a91c8
[1816890.744575] [c4499a58] [c01523cc] 0xc01523cc
[1816890.749011] [c4499a78] [c0152990] 0xc0152990
[1816890.753446] [c4499d88] [c0153610] 0xc0153610
[1816890.757881] [c4499ee8] [c01538d4] 0xc01538d4
[1816890.762317] [c4499f38] [c000f1c8] 0xc000f1c8
[1816890.766753] --- interrupt: c01 at 0xb7f669d4
[1816890.766753]     LR = 0xb7f53f30
[1816890.774484] Instruction dump:
[1816890.777619] 40820010 891f002d 71080040 40820054 7f9d3800 40bdfe44 39290003 4bfffde4 
[1816890.785540] 54c6073e 2b86000a 38c00001 7cc0371e <0f060000> 7f9d3800 40bdfe20 39290003 
[1816890.793635] ---[ end trace 65759db651e0e282 ]---
```

pktpls, 2021-12-09:
And on EAP225-Outdoor

spolack, 2021-12-09:
I'm now testing snapshot kernel with mac80211 backported from 5.15rc6 on magda-ap4
```
kmod-ath - 5.4.162+5.15-rc6-1-1
kmod-ath9k - 5.4.162+5.15-rc6-1-1
kmod-ath9k-common - 5.4.162+5.15-rc6-1-1
```

spolack, 2021-12-12:
> I'm now testing snapshot kernel with mac80211 backported from 5.15rc6 on magda-ap4
> 
> ```
> kmod-ath - 5.4.162+5.15-rc6-1-1
> kmod-ath9k - 5.4.162+5.15-rc6-1-1
> kmod-ath9k-common - 5.4.162+5.15-rc6-1-1
> ```

With that software versions the issue is fixed.

spolack, 2021-12-12:
@pktpls can you confirm?

spolack, 2021-12-24:
Its also fixed on 21.02-SNAPSHOT
```
root@magda-ap4:~# opkg list | grep ath
kmod-ath - 5.4.163+5.10.68-1-3
kmod-ath9k - 5.4.163+5.10.68-1-3
kmod-ath9k-common - 5.4.163+5.10.68-1-3
````

closed 2021-12-24 by spolack