freifunk-berlin/bbb-configs #117: ath10k_pci stacktraces: failed to send pdev bss chan info request

spolack, 2021-12-16:
On dorfplatz-ap3 (eap225-Outdoor) the wifi suddenly stops working + logging
```
[235813.427635] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235818.936355] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235819.436211] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235824.944839] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235825.025698] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235825.034982] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235825.055173] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235825.064559] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235825.444918] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235830.953283] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235831.453601] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235836.964725] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235837.463754] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235842.973176] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235843.472311] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235848.981552] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235849.480857] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235854.989918] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235855.025217] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235855.034432] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235855.054159] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235855.063848] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235855.490487] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235860.998296] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235861.500001] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235867.006675] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235867.508724] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235873.015034] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235873.517331] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235879.023941] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235879.525984] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235885.032527] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235885.045579] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235885.057825] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235885.084963] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235885.095082] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235885.536366] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235891.044926] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235891.544939] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235897.053298] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235897.553597] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235903.061692] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235903.562215] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235909.070221] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235909.570783] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235915.026245] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235915.036248] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235915.055822] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235915.065179] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235915.078620] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235915.579674] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235921.087029] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235921.588386] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235927.095727] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235927.597676] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235933.104238] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235933.606606] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235939.113448] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235939.616870] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235945.026164] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235945.035436] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235945.055244] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235945.064718] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235945.121920] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235945.626887] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235951.130292] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235951.635466] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235957.139075] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235957.644237] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235963.148790] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235963.652835] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235969.157170] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235969.661519] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235975.032474] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235975.042164] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235975.063542] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235975.073609] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235975.165583] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235975.670334] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235981.174033] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235981.678716] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235987.185241] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235987.688376] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235993.193681] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235993.696951] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235999.202231] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[235999.705504] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236005.029582] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236005.039353] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236005.060879] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236005.071040] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236005.210615] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236005.714062] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236011.219243] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236011.722733] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
[236017.227653] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
```

spolack, 2021-12-16:
```
root@dorfplatz-ap3:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02-SNAPSHOT'
DISTRIB_REVISION='r16401-14940aee45'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02-SNAPSHOT r16401-14940aee45'
DISTRIB_TAINTS=''
```

```
root@dorfplatz-ap3:~# opkg list | grep ath10k
ath10k-board-qca9888 - 20201118-3
ath10k-firmware-qca9888 - 20201118-3
kmod-ath10k - 5.4.163+5.10.68-1-3
```

PolynomialDivision, 2021-12-16:
OpenWrt suggested solution (for some other device with similar logs):
>  As of r8841-8c6662a1ac, ar71xx and ath79 have been switched to use ath10k-ct by default. This problem hasn't occurred for me with ath10k-ct. Please upgrade to a more recent snapshot. https://bugs.openwrt.org/index.php?do=details&task_id=333

spolack, 2021-12-16:
Seems to be no solution. ct is much more unstable in many scenarios.

PolynomialDivision, 2021-12-16:
> Seems to be no solution. ct is much more unstable in many scenarios.

I am also not happy. However, I think that was a reason why ct firmware is a thing? Qualcomm is not fixing their firmware, so that is why they switched to ct.

spolack, 2021-12-16:
Both, OpenWRT 21.02 and OpenWRT snapshot are running Firmware
`10.4-3.9.0.2-00131`

PolynomialDivision, 2021-12-16:
Awesome. Thanks!

spolack, 2021-12-17:
It crashed again. Now testing latest firmware `10.4-3.9.0.2-00156`

spolack, 2021-12-17:
```
root@dorfplatz-ap3:~# dmesg | grep ath10k_pci
[   11.789484] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   11.796268] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   14.305710] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
[   14.315282] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   14.335609] ath10k_pci 0000:00:00.0: firmware ver 10.4-3.9.0.2-00156 api 5 features no-p2p,mfp,peer-flow-ctrl,allows-mesh-bcast,no-ps,iram-recovery crc32 dd3f9ec1
[   14.663445] ath10k_pci 0000:00:00.0: board_file api 2 bmi_id 0:24 crc32 f228337a
[   16.690954] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   38.259416] ath10k_pci 0000:00:00.0: Unknown eventid: 36933
```

Now running.

PolynomialDivision, 2021-12-17:
Here is a patch that applies the small buffer changes, please copy to `openwrt/package/kernel/mac80211/patches/ath10k/999-ath10k-small-buffers.patch`
```diff
--- a/drivers/net/wireless/ath/ath10k/htt.h
+++ b/drivers/net/wireless/ath/ath10k/htt.h
@@ -235,7 +235,7 @@ enum htt_rx_ring_flags {
 };
 
 #define HTT_RX_RING_SIZE_MIN 128
-#define HTT_RX_RING_SIZE_MAX 2048
+#define HTT_RX_RING_SIZE_MAX 512
 #define HTT_RX_RING_SIZE HTT_RX_RING_SIZE_MAX
 #define HTT_RX_RING_FILL_LEVEL (((HTT_RX_RING_SIZE) / 2) - 1)
 #define HTT_RX_RING_FILL_LEVEL_DUAL_MAC (HTT_RX_RING_SIZE - 1)
--- a/drivers/net/wireless/ath/ath10k/pci.c
+++ b/drivers/net/wireless/ath/ath10k/pci.c
@@ -131,7 +131,7 @@ static const struct ce_attr pci_host_ce_
 		.flags = CE_ATTR_FLAGS,
 		.src_nentries = 0,
 		.src_sz_max = 2048,
-		.dest_nentries = 512,
+		.dest_nentries = 128,
 		.recv_cb = ath10k_pci_htt_htc_rx_cb,
 	},
 
@@ -140,7 +140,7 @@ static const struct ce_attr pci_host_ce_
 		.flags = CE_ATTR_FLAGS,
 		.src_nentries = 0,
 		.src_sz_max = 2048,
-		.dest_nentries = 128,
+		.dest_nentries = 64,
 		.recv_cb = ath10k_pci_htc_rx_cb,
 	},
 
@@ -167,7 +167,7 @@ static const struct ce_attr pci_host_ce_
 		.flags = CE_ATTR_FLAGS,
 		.src_nentries = 0,
 		.src_sz_max = 512,
-		.dest_nentries = 512,
+		.dest_nentries = 128,
 		.recv_cb = ath10k_pci_htt_rx_cb,
 	},
 
@@ -192,7 +192,7 @@ static const struct ce_attr pci_host_ce_
 		.flags = CE_ATTR_FLAGS,
 		.src_nentries = 0,
 		.src_sz_max = 2048,
-		.dest_nentries = 128,
+		.dest_nentries = 96,
 		.recv_cb = ath10k_pci_pktlog_rx_cb,
 	},
```

If something is missing, do

    make package/mac80211/{clean,prepare} .j1 V=s QUILT=1
    
go to (similar path for other devices and kernel versions):

     openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/backports-5.15.8-1/

now apply patches

    quilt push ath10k/999-ath10k-small-buffers.patch

Now you can use quilt edit [FILE] to appply further changes.

Use

    quilt refresh
    
to update patches. Afterwards copy the patch back to

    openwrt/package/kernel/mac80211/patches/ath10k

Don't forget to do another

    make package/mac80211/clean
    
Before building a new image.

spolack, 2021-12-24:
> It crashed again. Now testing latest firmware `10.4-3.9.0.2-00156`

Turns out latest firmware seems much more stable in our usecase. dorfplatz-ap3 is running stable for almost 6 days without cras

http://monitor.berlin.freifunk.net/host.php?h=dorfplatz-ap3&p=memory

So a new firmware seems to solve that issue on low-utilized APs. A proper fix would be still required for more dense environments.

We need to wait until ath10k-smallbuffers has made it upstream.
https://github.com/openwrt/openwrt/pull/4865

Noki, 2022-05-11:
I still have similar issues with one EAP225 that are not fixed in the smullbuffers variant and might be related and are not fixed in the latest builds: https://github.com/openwrt/openwrt/issues/9817

pktpls, 2022-05-27:
huette-core EAP225-Outdoor

```
FREIFUNK_DISTRIB_ID="Freifunk Falter"
FREIFUNK_RELEASE='1.2.2-snapshot'
FREIFUNK_OPENWRT_BASE='21.02-SNAPSHOT'
FREIFUNK_REVISION='eec1e53'
```

Syslog:
```
...
Fri May 27 16:48:33 2022 daemon.err babeld[2874]: send: Resource temporarily unavailable
Fri May 27 16:48:34 2022 daemon.err babeld[2874]: send: Resource temporarily unavailable
Fri May 27 16:48:34 2022 daemon.err olsrd[2808]: OLSR: sendto IPv4 'Resource temporarily unavailable' on interface wlan5-mesh
Fri May 27 16:48:36 2022 daemon.err olsrd[2808]: OLSR: sendto IPv4 'Resource temporarily unavailable' on interface wlan5-mesh
...
```

<details><summary>Kernel log >></summary>
<pre>
[    0.000000] Linux version 5.4.188 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16554-1d4dea6d4f)) #0 Sat Apr 16 12:59:34 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link EAP225-Outdoor v1
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 122184K/131072K available (5259K kernel code, 192K rwdata, 688K rodata, 1212K init, 205K bss, 8888K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] random: get_random_bytes called from 0x80661a28 with crng_init=0
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000006] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008228] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
[    0.046712] pid_max: default: 32768 minimum: 301
[    0.051715] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.059433] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.071527] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.081842] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.089121] pinctrl core: initialized pinctrl subsystem
[    0.097026] NET: Registered protocol family 16
[    0.127645] clocksource: Switched to clocksource MIPS
[    0.133911] thermal_sys: Registered thermal governor 'step_wise'
[    0.134258] NET: Registered protocol family 2
[    0.145368] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.153726] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.162611] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.170700] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... 467 lines cut ...]
[2054922.319436] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[2063048.339733] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2065843.181243] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[2070680.184133] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[2071409.387304] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[2130662.203472] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[2148482.364780] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[2197834.356656] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2204554.300165] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2217897.235262] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[2227178.238078] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2228650.327067] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[2229611.386067] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2232202.344432] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[2232993.294150] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[2233823.383751] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[... cut ...]
[2269903.220689] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2271946.325653] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2297439.307280] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[2302395.318033] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[2303886.321481] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[2305691.042295] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
<pre>
</summary>

Noki, 2022-09-11:
All my EAP225 devices are stable for weeks now. I think the problems are solved and this issue can be closed. @spolack can you confirm that this issue is obsolete?

pktpls, 2022-09-23:
On a very recent 22.03-snapshot, huette-core's wifi has completely locked up. First time I'm seeing it this bad :-(
```
> cat /etc/freifunk_release 
FREIFUNK_DISTRIB_ID="Freifunk Falter"
FREIFUNK_RELEASE='1.3.0-snapshot'
FREIFUNK_OPENWRT_BASE='22.03-SNAPSHOT'
FREIFUNK_REVISION='8112baf'
```
```
Fri Sep 23 19:05:13 2022 kern.warn kernel: [570667.232527] ath10k_pci 0000:00:00.0: wmi command 37004 timeout, restarting hardware
Fri Sep 23 19:05:13 2022 kern.warn kernel: [570667.240569] ath10k_pci 0000:00:00.0: already restarting
Fri Sep 23 19:05:13 2022 kern.warn kernel: [570667.246092] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
Fri Sep 23 19:05:16 2022 kern.warn kernel: [570670.272519] ath10k_pci 0000:00:00.0: wmi command 37004 timeout, restarting hardware
Fri Sep 23 19:05:16 2022 kern.warn kernel: [570670.280615] ath10k_pci 0000:00:00.0: already restarting
Fri Sep 23 19:05:16 2022 kern.warn kernel: [570670.286178] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
Fri Sep 23 19:05:19 2022 kern.warn kernel: [570673.312510] ath10k_pci 0000:00:00.0: wmi command 37004 timeout, restarting hardware
Fri Sep 23 19:05:19 2022 kern.warn kernel: [570673.320614] ath10k_pci 0000:00:00.0: already restarting
Fri Sep 23 19:05:19 2022 kern.warn kernel: [570673.326174] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
Fri Sep 23 19:05:19 2022 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Fri Sep 23 19:05:22 2022 kern.warn kernel: [570676.352512] ath10k_pci 0000:00:00.0: wmi command 37004 timeout, restarting hardware
Fri Sep 23 19:05:22 2022 kern.warn kernel: [570676.360518] ath10k_pci 0000:00:00.0: already restarting
Fri Sep 23 19:05:22 2022 kern.warn kernel: [570676.366047] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request
Fri Sep 23 19:05:22 2022 daemon.err collectd[2979]: plugin_read_thread: read-function of the `iwinfo' plugin took 44.064 seconds, which is above its read interval (30.000 seconds). You might want to adjust the `Interval' or `ReadThreads' settings.
```

Noki, 2022-11-29:
should be fixed by #379. reopen if needed.

closed 2022-11-29 by Noki