freifunk-berlin/bbb-configs #135: When uplink connectivity is broken without interface going down IPv4 mesh<->mesh traffic might get blackholed

spolack, 2022-02-06:
This is due to a static route which will be installed unconditionally.

https://github.com/freifunk-berlin/bbb-configs/blob/39907ea071d9e029962b6889f9dcae9a2cc8524f/roles/cfg_openwrt/templates/uplink_gateway/config/network.j2#L15-L19

Two ways to fix it:
* Have IPv4 BGP Session to ISP, making them only advertise a default route to us
* Have some ping-check mechanism which manages the default route weather a certain IP is reachable or not.

PolynomialDivision, 2022-02-06:
> Have IPv4 BGP Session to ISP, making them only advertise a default route to us

Sounds awesome.

Akira25, 2022-02-08:
> Have IPv4 BGP Session to ISP, making them only advertise a default route to us
Sounds complicated. The ping-solution might be understood even without detailed knowledge of a further protocol.

PolynomialDivision, 2022-02-08:
> Sounds complicated. The ping-solution might be understood even without detailed knowledge of a further protocol.

With ping, we rely on operators not violating any end-to-end principle or whatever. Here is a great example of people doing so:
https://twitter.com/DreselChristian/status/1490610345597190148

The config will probably look like this:
```
protocol bgp {
	local 10.0.0.1 as 44194;
	neighbor 10.0.0.2 as XXXX;
	ipv4 {
		import filter default_route;
	};
}
```
I guess the big ISPs have better mechanisms to check if they can advertise us or their partners a default route, then we could probably implement. We have BGP running on the gateways so the setup will not be a big change. I also prefer this solution, because if an ISP can stop announcing the default route, before killing our network or whatever. So they can do some "smooth" stopping of the default route (however, not sure if something like that is done). I like the idea of doing more cooperation between Transit networks and our by establishing even for IPv4 a BGP session. Maybe this will give us new and interesting opportunities in the future. We can go back to the ping solution whenever we want, however, BGP needs a running service also on the transit/isp part. I would prefer enabling a BGP sessions on all transit links, and if that is stupid, we can just going back. That is the "backward compatible" option.

spolack, 2022-02-08:
Usually ISPs are withdrawing all the routes ("called draining") before maintenance to gracefully shift the traffic away, therefore that approach is technically superior.

We anyway use bgp for the IPv6 routing, therefore i feel it would not really introduce "another protocol" or technology to learn,

Akira25, 2022-02-08:
Okay. That are good reasons to do so. Then let's do it that way.