freifunk-berlin/bbb-configs #185: device name length can exceed 15 chars

PolynomialDivision, 2022-03-29:
Bridge-interfaces are are prefixed with "br-", so device can exceed maximum lengths even if they are just 15 chars.

Noki, 2023-08-14:
I ran into this today... - again. How can we make the config and image generation when this happens?

Noki, 2023-08-14:
We have multiple interface names in the repo that are too long and therefore exceed the 15 char limit of the linux kernel. We should really detect this and fail with a notice:

mesh_perleberger36
mesh_richardplatz
mesh_whateverbla

Noki, 2023-08-15:
The length check from https://github.com/freifunk-berlin/bbb-configs/commit/81ee7086a96d5abd1d85234a63e7feb8b7c9d824 needs to be added here as well:

https://github.com/freifunk-berlin/bbb-configs/blob/0d82ff885eb4c9f7153f8245978c4d4702fc5c51/roles/cfg_openwrt/templates/common/config/network.j2#L34
https://github.com/freifunk-berlin/bbb-configs/blob/0d82ff885eb4c9f7153f8245978c4d4702fc5c51/roles/cfg_openwrt/templates/common/config/network.j2#L73
https://github.com/freifunk-berlin/bbb-configs/blob/0d82ff885eb4c9f7153f8245978c4d4702fc5c51/roles/cfg_openwrt/templates/corerouter/config/tunnelmanager.j2#L5

In addition we should probably change the text to be more specific:

"Network name is longer than 15 characters" => "Generated inteface name exceeds the 15 characters limit of the linux kernel. Try to shorten the name to resolve this."

PR #527 (2023-08-15): templates: check 15 character limit for bridges
This commit adds the 15 character limit check for bridge interfaces and adjusts the condition to actually check for 15 characters.  This resolves #185. The 15 character limit is mentioned here: https://openwrt.org/docs/guide-user/network/tunneling_interface_protocols
closed 2023-08-15 by robertfoss