freifunk-berlin/bbb-configs #194: prometheus: failed to connect to hostapd

pktpls, 2022-04-18:
Getting a ton of these:

```
Mon Apr 18 16:05:09 2022 daemon.err prometheus-node-exporter-lua[16: Failed to connect to hostapd - wpa_ctrl_open: No such file or directory
Mon Apr 18 16:05:09 2022 daemon.info prometheus-node-exporter-lua[16: .../lua/prometheus-collectors/hostapd_ubus_stations.lua:62: attempt to index local 'clients_call' (a nil value)
```

cc @Noki

pktpls, 2022-04-21:
Wondering if hostapd has started to ignore mesh devices...

The causes for these two errors are:
1. `hostapd_cli -i wlan5-mesh all_sta` fails (in hostapd_stations.lua)
2. ubus `hostapd.wlan5-mesh` does not exist, so `get_clients` can't be called on it (in hostapd_ubus_stations.lua)

The interface `wlan5-mesh` exists, works fine and is listed in `ubus call network.wireless status`.

Noki, 2022-04-25:
see https://github.com/openwrt/packages/issues/18341 for more details

Noki, 2022-09-11:
Obsolete due to https://github.com/freifunk-berlin/bbb-configs/commit/4a231098708c7000156a2ac180e5599366f03248

closed 2022-09-11 by Noki