freifunk-berlin/bbb-configs #308: OLSRd wireguard connections cause bbb-traffic to go through (slow/low latency) DSL connections

pmelange, 2022-08-16:
The current implementation via wireguard tunnels connecting to multiple gw nodes causes bbb traffic to go through the private DSL connection (which can be slow and have a poor latency).

```
~$ mtr a.bbb-vpn.olsr --report
Start: 2022-08-16T11:58:20+0200
HOST: automaton                   Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- frei.funk                  0.0%    10    0.4   0.4   0.3   0.4   0.0
  2.|-- mid9.bht-core.olsr         0.0%    10   13.3   7.4   5.1  13.3   2.6
  3.|-- mid12.segen-core.olsr      0.0%    10   17.2  12.8  10.6  17.2   2.0
  4.|-- mid8.saarbruecker-gw.olsr  0.0%    10    9.5  12.7   9.5  17.4   2.5
  5.|-- newyorck-core.olsr         0.0%    10   51.7  53.3  43.0  76.9   9.3
  6.|-- mid3.ak36-gw.olsr          0.0%    10   87.2 112.6  66.5 256.1  52.9
  7.|-- a.bbb-vpn.olsr             0.0%    10   79.7 115.0  72.3 223.8  50.0
```

The best scenario is for the packets __not__ to go through a private tunnel-based connection twice (in this example once into newyorck and once out from newyorck) in order to reach another part the backbone.  The implementation on the falter firmware, which creates a tunneldigger interface to the bbb-vpn server prevents this kind of short-cut.  

In my opinion, the setup for bbb-configs is a very poor network model which need to be implemented differently and which I consider a bug.  I recommend that the bbb-configs team consider changing their OLSRd island connections to use the same implementation style as the falter firmware using tunneldigger.

@Noki @booo @robertfoss @spolack @rtznprmpftl @freifunkoperator @PolynomialDivision

pktpls, 2022-08-16:
Is Falter doing any additional steps, or does it prevent these DSL hops by simply connecting to not more than one tunnel servers?

pmelange, 2022-08-16:
> Is Falter doing any additional steps, or does it prevent these DSL hops by simply connecting to not more than one tunnel servers?

yes, it is because the falter firmware only connects to one bbb-vpn server.  

But the bbb-vpn server does some things transparently for all connected clients.  Most importantly, it prevents gateway HNA announcements (0.0.0.0/0) from other island nodes and also from the backbone from being sent to the clients.  The extra plugin for the OLSRd daemon can be found here (https://github.com/freifunk-berlin/external_olsrd/tree/drophna/lib/drophna).

Akira25, 2022-09-03:
Is there any chance, that any bbb-configs developer will comment on this? This issue decreases the performance in the Berlin Freifunk Network drastically.

spolack, 2022-09-05:
I've been on vacation during the summer, sorry for the delay.

It is not intended to prefer VPN mesh links over physical links. Unfortunately the traffic engineering capabilities using olsrd metric are limited, thats why its not possible to put a super high metric on the VPN links which can never beat the physical links, but still its quite unlikely, as we've put a LQM of 0.2 on all the VPN Links.

If a route via two consecutive ETX5 links wins, the physical links must be real bad which points us to the spot, where we can improve.

Currently the following route is active:
```
root@bht-core:~# mtr a.bbb-vpn.olsr --report
Start: 2022-09-05T22:12:37+0200
HOST: bht-core                    Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- mid15.segen-core.olsr      0.0%    10   26.5  21.7   4.8  49.1  12.6
  2.|-- mid7.saarbruecker-gw.olsr  0.0%    10   17.0  19.4   5.1  32.6   7.2
  3.|-- mid2.sama-core.olsr        0.0%    10    7.0  17.7   7.0  34.2   9.4
  4.|-- mid10.zwingli-core.olsr    0.0%    10    8.1  16.2   8.1  25.9   6.3
  5.|-- mid1.vonnegut.olsr         0.0%    10   27.0  21.9  13.4  27.5   5.1
  6.|-- mid7.emma-core.olsr        0.0%    10   29.8  26.9  14.8  41.0  10.2
  7.|-- mid8.rhnk-core.olsr        0.0%    10   39.5  28.7  20.4  39.5   6.9
  8.|-- mid10.rhxb-core.olsr       0.0%    10   27.3  32.5  25.7  40.1   4.9
  9.|-- mid8.dtmb-core.olsr        0.0%    10   35.6  37.7  28.3  61.8   9.6
 10.|-- mid9.ak36-gw.olsr          0.0%    10   30.9  33.7  24.3  50.0   8.0
 11.|-- a.bbb-vpn.olsr             0.0%    10   31.1  40.0  26.2  69.8  13.0
```

spolack, 2022-09-05:
> The best scenario is for the packets not to go through a private tunnel-based connection twice (in this example once into newyorck and once out from newyorck) in order to reach another part the backbone.

I agree. We could avoid that by olsrd-meshing our gateways via GRE tunnels, similary to what we do with IPv6. Depending on the LQM this will work, as the GRE tunnel  always win over two VPNs metrics. The challenge would be to prefer physical links over the GRE Tunnels. Maybe a LQM of 0.15 would be a reasonable choice?

> In my opinion, the setup for bbb-configs is a very poor network model which need to be implemented differently and which I consider a bug. 

I strongly disagree. We just enabled OLSRD to provide also IPv4 Backbone + Internet connectivity, similar to what we have with IPv6.

Otherwise we'd have to run one tunnel technology for IPv4 and a different one for IPv6. Less complexity -> Less pain :)

Btw, the current implementation was discussed in a informal working group of interested freifunkers about two years ago. I even drew a diagram

https://viewer.diagrams.net/?highlight=0000ff&edit=_blank&layers=1&nav=1&title=Freifunk#R7VxZk%2BI4Ev41RGxvBIRvw2NBVc10THdP7bC7tfMosLA1ZVusD47%2B9ZuSJeOLwk1hYBvqADstp6RU5peptOSePgk2v0Ro6X2lDvZ7muJsevpjT9M029bhi1G2gqKYZkZxI%2BJkNHVHmJLvWBAVQU2Jg%2BNSwYRSPyHLMnFOwxDPkxINRRFdl4stqF%2BudYlcXCNM58ivU1%2BJk3iCqirK7sKvmLieqHpoigszNH9zI5qGor6epi%2F4T3Y5QJKXKB97yKHrAkl%2F6umTiNIkOwo2E%2Bwz4UqxZfc977matzvCYdLmhtX828Nf0z%2BTP%2F7x9u3bzA6nr%2FS5rxkZmxXyUyz7YaFg2dPH4SxmX728A8lWCg04w%2FjAyRh6tWTEuU9TqGi89kiCp0s0Z8Q16AzQvCTw4UzlzHhtOErwZm8%2F1Fw6oHaYBjiJtlBE3NAfivYIjTM0oXDr3fCNpHZ5hZHTZUeQUBk3570TGxwIyf2AFNUmIfpQ7dghKzh02eF0Gz%2F5eIVDeQ2qKly%2BKhlrZRFLaRYkbBgNEtY6k7BWk%2FDviYcjIH2evlyV6IyKemoXl53VRjvHv7zs08ta2X9SIXf9AQ4ewhBAEATKOvBsDFkXXDfCLkoA19vyfI5oUOT6B55jsmJMH%2FECpVAcaDRNsLx3FlW5HbInB3BfnNIo8ahLQ%2BQ%2F7ahjDuaYiVKBs12ZL5QuhYL8hZNkK5wYShNaVh%2B8Icl%2F2O0DU5z9KZix48dN8WRbOHnBEYGxBn0WtBDGPWOkD%2BU5ZzUYKaYk7Pjxs23xrMoxTlCUPDB3yY0BxTGZS%2FIz8WUHfDTDPvTcwdGE%2BjTiktMV%2FpPLlAnyfeMAudM0muN3lFL4HajfxckhaN3vC5SBYRole%2BsLbxBhHyVMg0oBQ4NtceYgGrQtFFhSEiZxoe4XRthVrMooR1q5UXG%2B1fJD%2B73ycJC1oHK37FOl%2BXSxiEFqVWDI5fMBTza8HrtRP2I3ZslsOjSZS9iE3WwUrVX%2BQ%2B6kHjHGZLNGW7WmOcKI4Ngcwx8bFG2i8H8Trk%2BAprIzQZe07Fwt0xrLqQanyO%2FClaFpvnOJX5FFGvgNd1fgTx%2BzgIDAlOELg8cXGpOE0BD6NaNJAm6rpMuy7INPXFYmYTYwRvEym7wsyIZZyhiJy3PQEqZyYy4sHD1BgMhlphail2DjstnXYE7iOVVHg4jzGi%2Bjz3Pejog5RsZkAXpZAO7nB%2FbLlTaib7gE6aYCU7eTRDyqXo54dKse8pgNEY%2FZVcBzRRj2Id9fcf3D%2Fw8Usz%2Fm2c8DYqN7TCztgoNaWZdr4FQFtYA4TmY2OCbf0YzzUySK5ZD%2FeDikjGiCBJruIswGOBLpH1FVL58tFTV0Pxi8F0CatmqVI0jtmAiyFvJJNjlXc6CNzhXHadr1YODNxXFtEbCQBrwABNp74ri65tzjuJuI43TzyuI4Tb8IiF0AfE4FnCcEMVVvi2IXDeTUptz7Hb9uEb9so4xfhtoVMKnG9URXH5phmppdnmOOrBwCrzvGag9PF0Un8z7P%2FPnmmQdgSA66DKNUZaAo2shQVdsybMW2ulK2ekSPgwDxTkX47hVv0yuqlWf5ZsMD6fNG9UpNTT0n7t919HZ1tDrzvLyO1mcVKOg7aO7d9fSG9dSyy67dbFgYdV49bbW25yGkYrHUVxx77YO4hkARk0UavvV%2F60303tha4pDM346PCi%2B4TMsclofSGtaHUpeLPIpjqVqdDWY9flsT301V5Q46Nww6ul12jtalnaOuNYBORTfnabTiuY4sneDUcwMNiQ15nKUiZJLjiMRGnsMocPrRlIaDYi%2FvwfnzG1rbh0h71gK1VI4PPuM0KoGbZVSULuunuGundzVGplKJAEcVRpkcaox%2BdN1etR7rwLq9agdluxrX7Z3q4a2ut%2FHq%2F3751tpxv5IIuymKoNbn60nZ7AXHox%2F263uyfYK9MlA0Qy0NqHwMeqwhbBtv6O7Bvl5ffvc99VAa3%2BODG40PhpUEj63X98OcNz5oSnqfIj4QSlMMEpjK3HKYkD0GuZT7t83RYFT4qUypjg0GbP09tvrwo6HByTG51Sz87q%2F3QcVP7a%2BbVppWhk76mYWPNwI9mjFxD750OZVSToCRJ8S7tlskLjor0ioL4g3lSCDUKnsJ9arPPtGsSLMq9dhnmOVI7%2FA%2Ban7ROG4qPDqTe8BBATyaul5rPK3tvb3GJ9gdwOvoXXjtKwNdHw7L%2BKqdxAR0bTAq73jT9YFVefLdHepKp3HLqFuKTNtD8OF9GxcNOfVRJTasBofmcVBrWu%2FHnBW2p0pHVYHXOgfw1vO3U77yu%2F%2B5ALJ9jrs%2FAwoaB%2FcQnw8FtXOioFrPI2ZL%2FOvrqu%2B5ot4t5IrsSvpbl0HYpXJFRlOq%2B54r6iRXpLedPF3Www8tbWDbRu6Jy5vzbEMbaEf6%2BK7zSnv6U1mWbon86N5Wjsrl5UqFbmOCxvdA3VNYLaKLPcby86SwjAOL2BU5QED9b8peZzb28SLZneWzcOLgkLKV6WyZ%2BUE9as3371D2Cc09%2BJIrlOBwmqDQoRGTmAdCAnY9vkretLqv%2BzeCv8PX316R578BOf5UbQUzuCPWYf1Qu7Jgj1UaYyyqXqR8wFcEr%2BGLLuAj4IvCFBaZkNDlDoJZAZQPHfbJEy687Z%2FZRgUF4iP0xpaTnba1MxRjVh%2BESSy%2Fg5JqayelLqGIN1HsdhAN97DoCNv9wHJGVEh7GO%2F6s9szwS%2F%2FawkhF0YBd3CJJ%2BQUYIegMO8zsyHwz3BLhMIY7JY3VVTw9emRtRskojxM4eMFcTYvEV5CuMKb1tnAejT1WVNwKAZNbNRQnn20hK9HFLBVelw8Wf9jAASmjdsY0JRkcspll%2FeT0%2FhuCiknsFxGAuxce9gP8lv4pDFk0CPlB%2B64J9caKhMPhS4fDrD6%2FqfuRBGlIe8jfGQpwymOVoSP84KPzcQnPGAHu%2Fx9yaYFyM%2BUapayxodkpxMeWu3uE64hk6EwFhL7OD5pZ%2Fr9nnzxouCifGaVsAbwTu2SnyuCsk4OGPhsSCzaVi6P%2FLjQlUxOcJAufcIxSlhDLq6ss3K%2FT0J9DKo%2Bx513ktsX18N4CTOfUCiZbFfHIFl71%2BVDnGk2h5eAR7egIfwkZW8l5UYmJU3CGHyaEGMMo%2BAXt2IBF5yL91caZ7onNPYT5xjn3IsJbg4u2Sh%2F0SS0ZON9us5373%2BmdZhy8NKnWyGwZwgR14gjfELXECrGFTj2CHesNPT5LT572SuDl2TGX7vK0XixYHjVqYq%2BstXIPsnGL%2BBWxIcMRmtB3DTCTm5OAYJxhH%2FhHkRHXcKGGdwBy0dQFtR1Kfrb4F6ZOxxYMb43qcObcZLcyqiyVLDfsE53aDXMm4dHrCiH092bg7NAf%2Fd%2BZv3pfw%3D%3D

pktpls, 2022-09-05:
I think one easy option to improve the routing-through-islands situation is to only create one Wireguard tunnel on these nodes - currently we usually create two. The positive effect of having a secondary tunnel for failover is debatable, since Wireguard tunnels are super-super-fast to initialize and shake hands. We would even save bandwidth if there's one less OLSR interface on these nodes.

From what I can tell, this issue has been observed with wuhli, kts13, newyorck.

spolack, 2022-09-05:
Of course this is an option but still wouldnt solve the issue, that traffic *could* go through a tunnel, when being used as backup for a site, which is also physically connected to the BBB:

Akira25, 2022-09-06:
> I strongly disagree. We just enabled OLSRD to provide also IPv4 Backbone + Internet connectivity, similar to what we have with IPv6.
> 
That's one part of the problem. When bbb-vpn was introduced, it was part of the concept, to have separate tunnel connections for backbone-mesh and for internet-uplink, to circumvent the problems introduced with a combined tunnel. It is way more flexible to have those tunnel types separate.

Take this for example:
[image]

Applying the wirguard-tunnel paradigm of bbb-configs, router 4 must have an additional uplink to the internet via an tunnel, to get connectivity to the backbone for the island. But the island has already Internet uplink via router 5. Requiring tunnels to have both uplink and mesh is not making things easier, but introduces a lot more complexity in the whole setup. It requires much more complex metrics to get the desired effects in that situation.

> Otherwise we'd have to run one tunnel technology for IPv4 and a different one for IPv6. Less complexity -> Less pain :)
> 
I disagree. Having wireguard as additional tunnel technology, wasn't a decision in agreement with the community, but was just done. I would even question the benefits of it in our appliance. Tunneldigger (which is effectively l2tp) works perfectly since years, without any issues.

In addition I consider the wireguard-management stuff (the shell scripts) far away from being ready for production. We have hard-coded passwords in that code! It blows up quite often and is not well documented.

That being said, I consider it far less complex to deploy well tested code instead of having half-backed scripts potter around one core functionality of our network.

One could argue as well, that the complexity we save with having both uplink and mesh on the same tunnel is just moved on to more complexity in the routing algorithms.

> Btw, the current implementation was discussed in a informal working group of interested freifunkers about two years ago. I even drew a [...]

It's great to have a discussion with some few Freifunkers. But for deploying a technology in the whole net, as we have seen with wireguard, I'd say it should take more, than just having an 'informal discussion'.

spolack, 2022-09-06:
Arguing that the tunneldigger based gateways work perfectly fine for years must be a joke! Furthermore they are not really capable of working with the current IPv6 network architecture.

I'm glad that you mentioned the lack of governance and decision making process in freifunk berlin. It was an informal meeting with the outcome that we would like to test it. That has been made transparent (also to the mailing list) and the architecture worked like a charm. Havent seen anyone dropping constructive objections in that regard. For the sake of preserving history :)

Furthermore the architecture is more or less a continuation of what began many years ago with the introduction of IPv6 at freifunk Berlin. Unfortunately it stalled and became abandoned, but havent you never asked yourself why almost every location already had a public IPv6 prefix? :)

pmelange, 2022-09-06:
> Its not possible to apply that concept to proper IPv6 Public connectivity. You have to tunnel yourself into the mesh, there is no alternative, because home users rarely speak BGP with their ISP.

Totally wrong.  The l2tp tunnel interface can carry __any__ ethernet packet.    But I've said this before and it fell of deaf ears.  Anyhow, the wireguard implementation is also doing an ipv4 tunnel with ipv6 traffic in it (also just ethernet packets).  The big difference is the latency of encrypting the wireguard packets.  The tunneldigger uplink servers are just simply configured to do a basic NAT ipv4 uplink, but there is no reason they couldn't provide a babel interface for ipv6.

> I'm glad that you mentioned the lack of governance and decision making process in freifunk berlin. It was an informal meeting where all interested parties were invited to, with the outcome that we would like to test it. That has been made transparent (also to the mailing list) and the architecture worked like a charm. Havent seen anyone dropping constructive objections in that regard. I only wanted to mention that for the sake of preserving history :)

Excuse me, but I find that hard to swallow.  When I mentioned many months ago that something wasn't working right, @spolack's response was something like _it's new new standard, deal with it_.  The issue then was with a 20MBiit DSL line attached node (node A) using wireguard announcing to it's neighbors (nodes B, C, D etc) that it has a 1GBit uplink as a direct neighbor instead of node A announcing it's actual DSL line speed.  At the same time, an actual node (node E) is using the falter firmware and announces it's actual line speed of 500MBit (now image that node E's operator is willing to provide access without a tunnel).  Node A will take precedence.

But that is not the point of this issue.  There is a real-world problem.  What could be the solution?   Perhaps each node can be set up to have only __one__ wireguard olsr interface while having __N__ babel interfaces.

spolack, 2022-09-06:
I dont see how the transport protocol matters for the issue pointed out initially. We've choosen wireguard because its a stupid simple layer3 tunneling protocol. The encryption is a nice to have.

>> Its not possible to apply that concept to proper IPv6 Public connectivity. You have to tunnel yourself into the mesh, there is no alternative, because home users rarely speak BGP with their ISP.
> Totally wrong.

Could you please explain your allegation my statement regarding the current IPv6 architecture is wrong?

> When I mentioned many months ago that something wasn't working right, @spolack's response was something like it's the new standard, deal with it

Please proof your accusation! From the first moment when you raised that issue (shortcut via Wireguard VPN) i have acknowledged and promised to fix, but hardly havent had time to fix it right away, because we've just turned up a new site. Eventually nick implemented my idea how to fix that issue: https://github.com/freifunk-berlin/bbb-configs/commit/915c331867a42ae3c39b208b90cd9db76ed97a5d

pmelange, 2022-09-06:
> Please proof your accusation!

Off topic: It was in Matrix, and it was the last thing I read before I quit using matrix chat.

About the ipv6 stuff, it's also off topic.  But anyhow, the argument that the tunneldigger servers are not capable of working with the current IPv6 network architecture is wrong.  It's possible to configure them to work with GRE tunnels to the -gw nodes and the current babeld setup.  They are capable, just not configured.

```
root@nodeE:~# ping6 ff02::1%bbbdigger
PING ff02::1%bbbdigger (ff02::1%89): 56 data bytes
64 bytes from fe80::b460:1cff:fe9c:7a09: seq=0 ttl=64 time=1.096 ms
64 bytes from fe80::fca4:9bff:fe1b:5bbd: seq=0 ttl=64 time=9.760 ms (DUP!)
64 bytes from fe80::b460:1cff:fe9c:7a09: seq=1 ttl=64 time=0.620 ms
64 bytes from fe80::fca4:9bff:fe1b:5bbd: seq=1 ttl=64 time=10.028 ms (DUP!)
^C
--- ff02::1%bbbdigger ping statistics ---
2 packets transmitted, 2 packets received, 2 duplicates, 0% packet loss
round-trip min/avg/max = 0.620/5.376/10.028 ms
root@nodeE:~# ssh root@fe80::fca4:9bff:fe1b:5bbd%bbbdigger -p 2222

Host 'fe80::fca4:9bff:fe1b:5bbd%bbbdigger' is not in the trusted hosts file.
(ssh-rsa fingerprint sha1!! b2:b9:71:4a:19:63:1d:25:bb:c5:38:68:93:27:be:01:4c:1b:32:48)
Do you want to continue connecting? (y/n) y

BusyBox v1.30.1 () built-in shell (ash)

  _____        _  __             _
 |  ___|      (_)/ _|           | |
 | |_ _ __ ___ _| |_ _   _ _ __ | | __
 |  _| '__/ _ \ |  _| | | | '_ \| |/ /
 | | | | |  __/ | | | |_| | | | |   <
 \_| |_|  \___|_|_|  \__,_|_| |_|_|\_\

 Firmware Freifunk Berlin (Dev-daily rev 297be42)
 Generic - ramips/mt7621
 https://wiki.freifunk.net/Berlin:Firmware
 https://github.com/freifunk-berlin
 -----------------------------------------------------

 If you find bugs please report them at:

   https://github.com/freifunk-berlin/firmware/issues

 For questions write a mail to <berlin@berlin.freifunk.net>
 or check https://berlin.freifunk.net/contact for our weekly meetings.

root@a-bbb-vpn:~# 

```
So, tunneldigger is not the problem.  

> The encryption is a nice to have

Still off topic.  The encryption is a waste and causes high latency.  Even though a lot of web traffic nowadays is TLS end-to-end encrypted, let's pretend that it isn't.  What data is being protected on the wireguard side of the tcp connection or udp stream which someone can't sniff once outside of the wireguard tunnel (both on the mesh and in the internet)?  I hope the answer isn't about protecting ip addresses because they are almost all in this repo with geo locations and sometimes even names and email addresses.

Now, back on topic...

> [915c331](https://github.com/freifunk-berlin/bbb-configs/commit/915c331867a42ae3c39b208b90cd9db76ed97a5d)

https://github.com/freifunk-berlin/bbb-configs/issues/308#issue-1340122819 shows that it isn't the needed solution.

pmelange, 2022-09-06:
> Perhaps each node can be set up to have only **one** wireguard olsr interface while having **N** babel interfaces.

So, how about considering this?

Or: for ipv4 mesh connectivity use tunneldigger to the bbb-vpn server -AND- for an ipv4 uplink using tunneldigger to the uplink servers or possibly also the -gw nodes since the server side of tunneldigger (called the broker) does run on openWrt.

Akira25, 2022-09-07:
> Arguing that the tunneldigger based gateways work perfectly fine for years must be a joke! Furthermore they are not really capable of working with the current IPv6 network architecture.
> 
I'm not here for joking and you shouldn't be either. I've never had serious tunneldigger-dependend issues with the tunneldigger-gateways. Neither the bbb-config ones, nor the one of the community-tunnel. Getting blacklisted for misuse can happen with wireguard-tunnels too. But that blacklisting is clearly not the point in this issue. We talk on bbb-vpn connections. So please stop making those wrong implications on tunneldigger unless you can prove them.

> Could you please explain your allegation my statement regarding the current IPv6 architecture is wrong?

ehm... If I got it right, the explanation you're asking literally starts at the beginning of the next sentence, directly after your quote ends... In addition we might experience a misunderstanding here: As far as I understand, the 'totally wrong.' didn't correlate to the whole IPv6-Infrastructure, but to this statement:

> Its not possible to apply that concept to proper IPv6 Public connectivity. You have to tunnel yourself into the mesh, there is no alternative, because home users rarely speak BGP with their ISP.

Furthermore, as @pmelange, and @pktpls mentioned, there are alternatives to that. As far as I perceived it from the discussion so far, one problem maybe is, that you are not willing to examine them properly.

In addition, I clearly wouldn't enforce public IPv6-Adresses and connectivity, if it eliminates the idea no-hierarchy-mesh-networks, what Freifunk essentially is.

Maybe the whole bbb-configs stuff has a completely different intention here:

I'm not into building a small Internet like WISPs do. I consider internet connectivity only as one of many other services in a free, decentralised mesh network. Freifunk doesn't need to get part of the public IPv6 internet in a native at any price! We shouldn't abolish our main goal, which is wireless meshing, just for some public routable IPv6 adresses.

PolynomialDivision, 2022-09-08:
Voting for c

spolack, 2022-09-08:
IMO we should not vote but share our thoughts arguments in order to come to an agreement. I feel it should be both a) and c) in order to address the underlying issues in a proper manner.

Noki, 2022-09-09:
I want to contribute here:

We reenabled the DSL-uplink on k12-core by plugging the internet cable back in. The routing towards `a.bbb-vpn.olsr` on other nodes in the area than changes like this:

```
root@segen-core:~# mtr -4 -r a.bbb-vpn.olsr
Start: 2022-09-09T13:21:17+0200
HOST: segen-core                  Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- mid11.k12-core.olsr        0.0%    10    6.0   6.7   5.1   8.5   1.2
  2.|-- mid1.ak36-gw.olsr          0.0%    10   13.7  15.6  12.6  24.0   3.3
  3.|-- a.bbb-vpn.olsr             0.0%    10   16.1  15.3  12.8  19.2   2.1
```

```
root@e16-core:~# mtr -r a.bbb-vpn.olsr
Start: 2022-09-09T12:58:45+0200
HOST: e16-core                    Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- e16outdoor-core.olsr       0.0%    10    4.4   2.8   1.9   4.4   0.8
  2.|-- mid11.segen-core.olsr      0.0%    10   14.6  10.7   8.6  14.6   2.1
  3.|-- mid7.k12-core.olsr         0.0%    10   15.7  16.9  15.1  18.3   1.1
  4.|-- mid1.ak36-gw.olsr          0.0%    10   24.2  25.7  21.9  33.8   3.5
  5.|-- a.bbb-vpn.olsr             0.0%    10   35.6  29.4  22.9  35.6   3.9
```

I think this is exactly the unwanted behaviour that originally lead to this github issue.

spolack, 2022-09-09:
I can confirm, this is exactly what meant with

> Of course this is an option but still wouldnt solve the issue, that traffic could go through a tunnel, when being used as backup for a site, which is also physically connected to the BBB.

```
root@segen-core:~# mtr 10.230.38.204 --report
Start: 2022-09-09T14:04:33+0200
HOST: segen-core                  Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- mid11.k12-core.olsr        0.0%    10    5.7   6.3   5.1   7.8   0.9
  2.|-- mid1.ak36-gw.olsr          0.0%    10   14.7  15.3  13.0  19.7   2.3
  3.|-- a.bbb-vpn.olsr             0.0%    10   21.7  16.0  13.6  21.7   2.6
```

```
spolack:~$ ssh root@segen-core.olsr "echo /routes | nc 127.0.0.1 9090" | jq '.routes[] | select( .destination == "10.230.38.204" )'
{
  "destination": "10.230.38.204",
  "genmask": 32,
  "gateway": "10.31.158.133",
  "metric": 3,
  "etx": 7.099609,
  "rtpMetricCost": 7.099609,
  "networkInterface": "switch0.106"
```

Could you please provide the same information if you pull your DSL cable?

Currently one tunnel has the same cost/metric as 5 phyiscal hops.

Noki, 2022-09-09:
When disabling the uplink it looks as follows:

```
root@segen-core:~# mtr -4 -r a.bbb-vpn.olsr
Start: 2022-09-09T14:17:50+0200
HOST: segen-core                  Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- mid8.saarbruecker-gw.olsr  0.0%    10    1.0   1.0   0.9   1.5   0.2
  2.|-- mid2.sama-core.olsr        0.0%    10    2.0   2.0   1.6   2.2   0.2
  3.|-- mid10.zwingli-core.olsr    0.0%    10    2.8   3.0   2.3   3.8   0.4
  4.|-- mid1.vonnegut.olsr         0.0%    10    9.1   9.3   7.5  12.6   1.7
  5.|-- mid7.emma-core.olsr        0.0%    10   12.7  10.6   7.8  16.9   3.0
  6.|-- mid8.rhnk-core.olsr        0.0%    10   17.9  17.9  13.4  33.7   5.8
  7.|-- mid10.rhxb-core.olsr       0.0%    10   17.3  17.1  13.2  19.0   1.6
  8.|-- mid8.dtmb-core.olsr        0.0%    10   21.4  18.3  14.8  22.7   2.5
  9.|-- mid10.ak36-gw.olsr         0.0%    10   22.6  20.3  17.8  22.6   1.7
 10.|-- a.bbb-vpn.olsr             0.0%    10   19.3  20.2  18.2  22.8   1.4
```

```
root@e16-core:~# mtr -4 -r a.bbb-vpn.olsr
Start: 2022-09-09T14:18:20+0200
HOST: e16-core                    Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- e16outdoor-core.olsr       0.0%    10    4.8   3.5   1.8   5.2   1.2
  2.|-- mid11.segen-core.olsr      0.0%    10   13.0  11.8   9.0  15.5   1.9
  3.|-- mid7.saarbruecker-gw.olsr  0.0%    10   11.9  10.6   8.1  15.7   2.4
  4.|-- mid2.sama-core.olsr        0.0%    10   16.7  14.3  12.3  16.7   1.4
  5.|-- mid10.zwingli-core.olsr    0.0%    10   32.0  15.0  10.2  32.0   6.4
  6.|-- mid1.vonnegut.olsr         0.0%    10   17.7  21.2  15.2  28.2   4.1
  7.|-- mid7.emma-core.olsr        0.0%    10   20.4  22.1  18.9  27.8   2.6
  8.|-- mid8.rhnk-core.olsr        0.0%    10   48.4  29.0  24.1  48.4   7.1
  9.|-- mid10.rhxb-core.olsr       0.0%    10   24.1  26.3  24.1  28.4   1.6
 10.|-- mid8.dtmb-core.olsr        0.0%    10   33.7  30.6  25.7  35.3   3.1
 11.|-- mid10.ak36-gw.olsr         0.0%    10   44.8  32.7  25.2  44.8   5.5
 12.|-- a.bbb-vpn.olsr             0.0%    10   31.9  34.2  27.2  50.9   6.5
```

Feel free to enable/disable the "uplink" interface on k12-core at any time if you want to do more tests.

spolack, 2022-09-14:
Observations from Segen:
* without your VPN activated we have a ETX of 10.6
* with your VPN we have a ETX of 7. (5 VPN + 2 Hops)

Unfortunately it's not possible to make the LQM of the VPN Gateways to the clients lower than 0.2 (it will stop working below, i tested that. So you we can "compensate" maximum 5 Hops. But i've found a Trick. Since the OLSRD Metric is bidirectional (aka same metric for both directions), we can also apply a LQM from the other side. This would most likely multiply the ETX of that link.

LQM VPN Gateway (ak36-gw for example) 0.2 * LQM 1 VPN Client (k12-core for example)  = Aggregated LQM 0.2  = 5 Hops
LQM VPN Gateway 0.2 * LQM 0.5 VPN Client = Aggregated LQM 0.1  = 10 Hops
LQM VPN Gateway 0.2 * LQM 0.2 VPN Client = Aggregated LQM 0.04  = 25 Hops (thats the maximum cost we can "simulate" on a single link)

If you want to prevent segen from selecting your VPN over the current bbb link, you'd need to apply a LQM of 0.5 on your VPN side, which would result in a ETX of 10+2 = 12

```
spolack@spolack:~$ ssh root@segen-core.olsr "echo /routes | nc 127.0.0.1 9090" | jq '.routes[] | select( .destination == "10.230.38.204" )'
{
  "destination": "10.230.38.204",
  "genmask": 32,
  "gateway": "10.31.83.58",
  "metric": 10,
  "etx": 10.607422,
  "rtpMetricCost": 10.607422,
  "networkInterface": "switch0.101"
}
```

```
spolack@spolack:~$ mtr 10.230.38.204 --report
Start: 2022-09-14T20:54:40+0200
HOST: spolack                     Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- frei.funk                  0.0%    10    1.3   1.4   1.1   1.7   0.2
  2.|-- zwingli-core.olsr          0.0%    10    6.4   4.2   2.2   6.4   1.1
  3.|-- mid1.vonnegut.olsr         0.0%    10   13.4  11.1   8.5  16.1   2.6
  4.|-- mid7.emma-core.olsr        0.0%    10   10.6  11.0   8.1  14.0   1.6
  5.|-- mid8.rhnk-core.olsr        0.0%    10   15.4  20.0  13.4  31.9   6.4
  6.|-- mid10.rhxb-core.olsr       0.0%    10   15.8  18.6  14.7  23.8   3.0
  7.|-- mid8.dtmb-core.olsr        0.0%    10   19.4  18.6  15.2  21.4   2.2
  8.|-- mid9.ak36-gw.olsr          0.0%    10   22.3  22.7  19.7  25.9   2.2
  9.|-- a.bbb-vpn.olsr             0.0%    10   19.4  23.9  19.4  37.5   5.1
```

Noki, 2022-09-15:
@PolynomialDivision can you please give an example on how to set lqm on the uplink interfaces in tunnelmanager and document it in the corresponding section (https://github.com/freifunk-berlin/bbb-configs/blob/master/DEVELOPER.md#uplink-via-tunnel)? I think it would also be a good idea to introduce a LQM of 0.5 or even 0.2 on the client as default value.

Noki, 2022-09-29:
@PolynomialDivision ping!

Noki, 2022-10-01:
I think this is fixed when setting proper metrics for tunnelmanager as now documented with #334.

spolack, 2022-10-01:
Yep. At least change the default.

Noki, 2022-10-01:
@spolack change the example to 0.5 in the documentation.

PR #336 (2022-10-02): tunnelmanager: change default metric for client uplinks
This should fix #308 and once merged #334 will add this to the documentation.
closed 2022-10-02 by Noki