freifunk-berlin/bbb-configs #316: Error in rendering templates

Akira25, 2022-09-09:
When calling `ansible-playbook play.yml --limit=rhnk*` theres an error in rendering some configuration from the templates:

```ansible
TASK [cfg_openwrt : Render /etc templates] ********************************************************************************************************************************
skipping: [rhnk-ap1] => (item=config) 
skipping: [rhnk-ap1] => (item=collectd) 
skipping: [rhnk-ap1] => (item=dropbear) 
skipping: [rhnk-core] => (item=nftables.d) 
skipping: [rhnk-platz] => (item=config) 
skipping: [rhnk-core] => (item=config) 
skipping: [rhnk-platz] => (item=collectd) 
skipping: [rhnk-core] => (item=collectd) 
skipping: [rhnk-core] => (item=dropbear) 
skipping: [rhnk-platz] => (item=dropbear) 
changed: [rhnk-platz] => (item=sysupgrade.conf.j2)
changed: [rhnk-ap1] => (item=sysupgrade.conf.j2)
changed: [rhnk-core] => (item=sysupgrade.conf.j2)
changed: [rhnk-platz] => (item=sysctl.conf.j2)
changed: [rhnk-core] => (item=firewall.user.j2)
changed: [rhnk-ap1] => (item=sysctl.conf.j2)
changed: [rhnk-core] => (item=sysctl.conf.j2)
changed: [rhnk-platz] => (item=nftables.conf.j2)
changed: [rhnk-ap1] => (item=nftables.conf.j2)
changed: [rhnk-core] => (item=nftables.conf.j2)
changed: [rhnk-platz] => (item=rc.local.j2)
changed: [rhnk-ap1] => (item=rc.local.j2)
changed: [rhnk-core] => (item=rc.local.j2)
changed: [rhnk-platz] => (item=config/naywatch.j2)
changed: [rhnk-ap1] => (item=config/naywatch.j2)
changed: [rhnk-platz] => (item=config/irqbalance.j2)
changed: [rhnk-core] => (item=nftables.d/20-wg-maxseg-size.nft.j2)
changed: [rhnk-ap1] => (item=config/irqbalance.j2)
changed: [rhnk-core] => (item=config/naywatch.j2)
changed: [rhnk-platz] => (item=config/wireless.j2)
changed: [rhnk-ap1] => (item=config/wireless.j2)
changed: [rhnk-core] => (item=config/tunnelmanager.j2)
changed: [rhnk-platz] => (item=config/network.j2)
changed: [rhnk-ap1] => (item=config/network.j2)
changed: [rhnk-core] => (item=config/irqbalance.j2)
changed: [rhnk-ap1] => (item=config/system.j2)
skipping: [rhnk-ap1] => (item=collectd/conf.d) 
changed: [rhnk-platz] => (item=config/system.j2)
skipping: [rhnk-platz] => (item=collectd/conf.d) 
changed: [rhnk-ap1] => (item=collectd/conf.d/basic.conf.j2)
changed: [rhnk-platz] => (item=collectd/conf.d/basic.conf.j2)
changed: [rhnk-core] => (item=config/wireless.j2)
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ): '<' not supported between instances of 'list' and 'int'
failed: [rhnk-core] (item=config/olsrd.j2) => {
    "ansible_loop_var": "item",
    "changed": false,
    "item": {
        "ctime": 1662665125.5826063,
        "gid": 1000,
        "group": "martin",
        "mode": "0644",
        "mtime": 1662665125.5826063,
        "owner": "martin",
        "path": "config/olsrd.j2",
        "root": "/home/martin/Freifunk/dev/bbb-configs/roles/cfg_openwrt/templates/corerouter",
        "selevel": "s0",
        "serole": "object_r",
        "setype": "user_home_t",
        "seuser": "unconfined_u",
        "size": 2336,
        "src": "/home/martin/Freifunk/dev/bbb-configs/roles/cfg_openwrt/templates/corerouter/config/olsrd.j2",
        "state": "file",
        "uid": 1000
    }
}

[... 160 lines cut ...]
config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'
): unsupported operand type(s) for -: 'list' and 'int'
changed: [rhnk-ap1] => (item=dropbear/authorized_keys.j2)
changed: [rhnk-platz] => (item=dropbear/authorized_keys.j2)
changed: [rhnk-core] => (item=config/airos-dfs-reset.j2)
changed: [rhnk-core] => (item=config/network.j2)
changed: [rhnk-core] => (item=config/babeld.j2)
changed: [rhnk-core] => (item=config/system.j2)
changed: [rhnk-core] => (item=config/firewall.j2)
changed: [rhnk-core] => (item=config/freifunk.j2)
skipping: [rhnk-core] => (item=collectd/conf.d) 
changed: [rhnk-core] => (item=collectd/conf.d/basic.conf.j2)
changed: [rhnk-core] => (item=collectd/conf.d/bbb-snmp.conf.j2)
changed: [rhnk-core] => (item=dropbear/authorized_keys.j2)
```

This happens on other locations too.

After some try-and-error, I found that 6c54984 didn't contain that error. Bisecting from there gives 7615844b43f403488c012c2bcf32dc0570298bae as commit from where this issue started happening.

I'm not sure, how to fix this though and I cannot see why this is happening. @pktpls do you have an idea on this?

spolack, 2022-09-09:
I've already prepared a revert patch i carry local, because i havent really had time to dig deeper.

```
diff --git a/roles/cfg_openwrt/tasks/flash.yml b/roles/cfg_openwrt/tasks/flash.yml
index 2732959..c9b7b09 100644
--- a/roles/cfg_openwrt/tasks/flash.yml
+++ b/roles/cfg_openwrt/tasks/flash.yml
@@ -5,7 +5,7 @@
 
 - name: "Set ansible ssh target from IPv6 Prefix"
   set_fact:
-    ansible_ssh_host: "{{ ipv6_prefix | ansible.utils.ipsubnet('64', mgmt_network['ipv6_subprefix']) | ansible.utils.ipaddr(1) | ansible.utils.ipaddr('address') }}"
+    ansible_ssh_host: "{{ ipv6_prefix | ipsubnet('64', mgmt_network['ipv6_subprefix']) | ipaddr(1) | ipaddr('address') }}"
     ansible_user: root
 
 - name: "Transfer image to remote host:"
diff --git a/roles/cfg_openwrt/templates/common/config/network.j2 b/roles/cfg_openwrt/templates/common/config/network.j2
index 78bf3ef..90cd239 100644
--- a/roles/cfg_openwrt/templates/common/config/network.j2
+++ b/roles/cfg_openwrt/templates/common/config/network.j2
@@ -39,12 +39,12 @@ config interface '{{ name }}'
     {% endif %}
     {% if 'assignments' in network and inventory_hostname in network['assignments'] %}
        option proto 'static'
-       option ipaddr '{{ network['prefix'] | ansible.utils.ipaddr(network['assignments'][inventory_hostname]) }}'
+       option ipaddr '{{ network['prefix'] | ipaddr(network['assignments'][inventory_hostname]) }}'
        {% if role != "corerouter" and 'dns' in network %}
-       option dns '{{ network['prefix'] | ansible.utils.ipaddr(network['dns']) | ansible.utils.ipaddr('address') }}'
+       option dns '{{ network['prefix'] | ipaddr(network['dns']) | ipaddr('address') }}'
         {% endif %}
        {% if 'gateway' in network and network['assignments'][inventory_hostname] != network['gateway'] %}
-       option gateway '{{ network['prefix'] | ansible.utils.ipaddr(network['gateway']) | ansible.utils.ipaddr('address') }}'
+       option gateway '{{ network['prefix'] | ipaddr(network['gateway']) | ipaddr('address') }}'
         {% endif %}
         {% if role != 'corerouter' and 'ipv6_subprefix' in network %}
 
@@ -61,8 +61,8 @@ config interface '{{ name }}6'
   {% endif %}
   {% if role == 'corerouter' and ipv6_prefix is defined %}
     {% if 'ipv6_subprefix' in network %}
-      {% set subprefix = ipv6_prefix | ansible.utils.ipsubnet('64', network['ipv6_subprefix']) %}
-       option ip6addr '{{ subprefix | ansible.utils.ipaddr(1) | ansible.utils.ipaddr('address') }}/{{ '128' if network['role'] == 'mesh' else '64' }}'
+      {% set subprefix = ipv6_prefix | ipsubnet('64', network['ipv6_subprefix']) %}
+       option ip6addr '{{ subprefix | ipaddr(1) | ipaddr('address') }}/{{ '128' if network['role'] == 'mesh' else '64' }}'
     {% endif %}
   {% endif %}
 
diff --git a/roles/cfg_openwrt/templates/corerouter/config/babeld.j2 b/roles/cfg_openwrt/templates/corerouter/config/babeld.j2
index e763eaa..bd33a17 100644
--- a/roles/cfg_openwrt/templates/corerouter/config/babeld.j2
+++ b/roles/cfg_openwrt/templates/corerouter/config/babeld.j2
@@ -25,7 +25,7 @@ config filter
 config filter
        option 'type'   'redistribute'
        option 'ip'     '{{ ipv6_prefix }}'
-       option 'eq'     '{{ ipv6_prefix | ansible.utils.ipaddr('prefix') }}'
+       option 'eq'     '{{ ipv6_prefix | ipaddr('prefix') }}'
 
 config filter
        option 'type'   'redistribute'
diff --git a/roles/cfg_openwrt/templates/corerouter/config/dhcp.j2 b/roles/cfg_openwrt/templates/corerouter/config/dhcp.j2
index 9262010..b8b5aea 100644
--- a/roles/cfg_openwrt/templates/corerouter/config/dhcp.j2
+++ b/roles/cfg_openwrt/templates/corerouter/config/dhcp.j2
@@ -30,7 +30,7 @@ config dhcp 'dhcp_{{ name }}'
        option force '1'
        option leasetime '5m'
        option start '2'
-       option limit '{{ network['prefix'] | ansible.utils.ipaddr('size') -3 }}'
+       option limit '{{ network['prefix'] | ipaddr('size') -3 }}'
[... 165 lines cut ...]
+       option netmask '{{ mgmt['ipv4'] | ipaddr('netmask') }}'
+       option netaddr '{{ mgmt['ipv4'] | ipaddr('network') }}'
   {% endif %}
 {% endif %}
diff --git a/roles/cfg_openwrt/templates/uplink_gateway/conntrackd/conntrackd.conf.j2 b/roles/cfg_openwrt/templates/uplink_gateway/conntrackd/conntrackd.conf.j2
index 349e335..b828c7c 100644
--- a/roles/cfg_openwrt/templates/uplink_gateway/conntrackd/conntrackd.conf.j2
+++ b/roles/cfg_openwrt/templates/uplink_gateway/conntrackd/conntrackd.conf.j2
@@ -5,7 +5,7 @@ Sync {
     }
 
     UDP {
-        IPv4_address {{ uplink['ipv4'] | ansible.utils.ipaddr('address') }}
+        IPv4_address {{ uplink['ipv4'] | ipaddr('address') }}
         IPv4_Destination_Address 127.0.0.1
         Port {{ conntrackd_port }}
         Interface lo
diff --git a/roles/cfg_openwrt/templates/uplink_gateway/samplicator.conf.j2 b/roles/cfg_openwrt/templates/uplink_gateway/samplicator.conf.j2
index df33aa1..71292af 100644
--- a/roles/cfg_openwrt/templates/uplink_gateway/samplicator.conf.j2
+++ b/roles/cfg_openwrt/templates/uplink_gateway/samplicator.conf.j2
@@ -1 +1 @@
-127.0.0.1:{% for host in groups['role_uplink_gateway'] if host != inventory_hostname %} {{ hostvars[host]['uplink']['ipv4'] | ansible.utils.ipaddr('address')  }}/{{conntrackd_port }}{% endfor %}
+127.0.0.1:{% for host in groups['role_uplink_gateway'] if host != inventory_hostname %} {{ hostvars[host]['uplink']['ipv4'] | ipaddr('address')  }}/{{conntrackd_port }}{% endfor %}
```

Noki, 2022-09-09:
`ansible-playbook play.yml --limit=rhnk*` works for me just fine:

```
PLAY RECAP *********************************************************************************************************
rhnk-ap1                   : ok=24   changed=0    unreachable=0    failed=0    skipped=7    rescued=0    ignored=0
rhnk-ap2                   : ok=24   changed=0    unreachable=0    failed=0    skipped=7    rescued=0    ignored=0
rhnk-ap3                   : ok=24   changed=0    unreachable=0    failed=0    skipped=7    rescued=0    ignored=0
rhnk-core                  : ok=25   changed=0    unreachable=0    failed=0    skipped=6    rescued=0    ignored=0
rhnk-platz                 : ok=25   changed=0    unreachable=0    failed=0    skipped=6    rescued=0    ignored=0
```

Maybe it is an environment thing and you need to upgrade the requirements or we need to increase the version numbers of our requirements.

pktpls, 2022-09-09:
Something about type mismatches. Have never seen that error, even though for 7615844b43f403488c012c2bcf32dc0570298bae I did check a full config run: `ansible-playbook play.yml -l all`. Did another run just now, all fine :-)

I'm on Ansible 2.12.6 at the moment, which is what `pip install -r requirements.txt` gave me back in June. The latest seems to be 2.13.3.

Akira25, 2022-09-10:
I was on `ansible-core-2.12.3`. After doing an upgrade and being on `core 2.13.3` the Issue magically disappeared.

closed 2022-09-10 by Akira25