freifunk-berlin/bbb-configs #339: documentation: add information about how to add packages to individual location

Noki, 2022-10-02:
Instead of adding utils to all locations as proposed in #47 we should add documentation how to install packages for individual location.

pmelange, 2023-08-07:
Is that possible?  How?  I would like to use that.

Noki, 2023-08-08:
I think you can just add a yaml file to your location and add or remove packages similar to the model files:

```
location__packages__to_merge:
  - "-kmod-ath10k-ct -ath10k-firmware-qca988x-ct"
  - "kmod-ath10k ath10k-firmware-qca988x"
```

Or add to the host file:

```
host__packages__to_merge:
  - "-kmod-ath10k-ct -ath10k-firmware-qca988x-ct"
  - "kmod-ath10k ath10k-firmware-qca988x"
```

I did not test this...

pmelange, 2023-08-14:
I created a file "host_vars/testnode/packages.yml" with the following:
```
---

host__packages__to_merge:
  - "collectd-mod-cpu htop"
```
And those packages were added

p.s. but unfortunately collectd-mod-cpu is not configured :(

Petrella, 2023-12-30:
I guess every package in the stream available for the router model will be added successfully?

Petrella, 2023-12-30:
So i will make a first draft and you can test and comment on that.