freifunk-berlin/bbb-configs #355: strom-gw: ipv4 ping on WAN doesn't work

pmelange, 2022-10-19:
In-Berlin contacted us that strom-gw is no longer ping-rechable for their monitoring system.  The reason is that the firewall does not allow ping on WAN.  Should a firewall rule be added to allow ping in the WAN firewall group?

pmelange, 2024-01-11:
Is there any chance that ping can be allowed on the uplink interfaces please?

FFHener, 2024-01-31:
While researching something else i just came across this rule. It looks like it should resolve this Issue, however i didnt tested it so far. I changed the title as the ping doesnt work on all gateways afaict.

```
config rule
	option name "Allow-Ping"
	option src uplink
	option proto icmp
	option icmp_type echo-request
	option family ipv4
	option target ACCEPT
```

closed 2026-02-08 by pktpls