freifunk-berlin/bbb-configs #370: RFC: Add configs for prop. Antennas

Akira25, 2022-11-02:
On most locations, there are antennas which do transparent WDS-Bridges PtP and PtMP-Links. The config for theses devices should be added to bbb-configs too.

For Mikrotik-Devices running RouterOS, one can export scripts, that will regenerate the current antennas config, by using the `export` command.

It might be a good idea, to encrypt the scripts via ansible-vault using the core router password. One example can be seen in `fffw-rathaus`-branch

pktpls, 2022-11-03:
Same thing can be done for AirOS, it has a textfile containing the full config which can be fetched via SSH.

spolack, 2022-11-05:
Why not setup a [oxidized](https://github.com/ytti/oxidized) server, which carries out backups to a git repo on a daily basis? We might need to implement support for some devices, but this would be definitely a more universal approach.