freifunk-berlin/bbb-configs #398: OWE transition mode needs a migration path

pktpls, 2022-12-21:
In the past few weeks since the introduction of OWE transition mode, feedback of problems has increased. People can't connect because their devices reject the change in security mode. A relatively easy to explain workaround is to delete the wifi connection on the device, and then reconnect. It's not obvious to users though, so it depends on direct contacts to Freifunk maintainers to understand it. Most of the users will not understand that and only see a "broken" wifi.

The other day someone told me of a possible additional failure scenario that I haven't tested myself yet. Given your device already knows the SSID with OWE transition mode and works fine, and you try to connect to a non-OWE SSID somewhere else, the same rejection-because-change-in-security-mode applies. That person got into a loop where they had to repeatedly delete the wifi connection when they went to the other palce, and will continue to have to do that.

Although I haven't tested the additional case yet, it makes sense: if devices are so strict that they even reject an upgrade in security (plain -> OWE) then they should obviously also reject a downgrade (OWE -> plain).

I'm not sure what to suggest.

One option is to version the SSID - e.g. "berlin.freifunk.net 2022" if it has OWE transition mode enabled. Yes that'd break automatic association with the "old" SSID, but that's more or less broken anyway: we now regularly have the situation where only deleting the wifi connection helps.

PolynomialDivision, 2022-12-21:
Hmmm, but actually you are not connecting into "berlin.freifunk.net", instead you are connecting to "berlin.freifunk.net (OWE)" (with a hidden SSID). Do you have some infromation about the devices?

spolack, 2022-12-22:
Havent experienced any problems. What are the involved devices / software stacks?

FFHener, 2022-12-22:
I can provide one text that i just got from Fesev:

"Es liegt wohl an der Kombi zwischen meinem PC und dem Freifunk router (andere haben das Problem nicht). Aber dieses Problem habe ich nur bei Freifunk auf dem Platz und nicht bei anderen Freifunkroutern. Konkret ist das Problem das mir nur die Verbindung berlin.freifunk.net (wlp2s0) angezeigt wird, zu dem ich nicht verbinden kann mit der Fehlermeldung "connection kann nicht hinzugefügt werden. 802-11-wireless-security.key-mgmt: Der Zugriffspunkt unterstützt kein PSK aber die Einstellung erfordert es.""

Regarding Lars text I told the person to just remove the Connection completle and than try to reconnect. I asked for more infos about the device as well, but it is an old ThinkPad running Linux.

noxilixon, 2023-02-13:
I was able to determine the following with my laptop, when I failed to connect with the _berlin.freifunk.net_ ssid. The _berlin.freifunk.net (OWE)_ ssid was visible at first, but disappeared from the list after some time.

Device used:
```
> lspci -nn | grep -i network
03:00.0 Network controller [0280]: Intel Corporation Wireless 7265 [8086:095b] (rev 59)
```

Journal (only one entry) :
```
> journalctl
Feb 13 10:40:28 rechenapparat gnome-shell[1662]: An active wireless connection, in infrastructure mode, involves no access point?
```

pktpls, 2023-03-07:
This turned out to be a completely different problem: see #425

I'll keep track of the bug in new issue #426 and close this one here since it's full of wrong ideas :)

closed 2023-03-07 by pktpls