freifunk-berlin/bbb-configs #416: Lint errors after ansible update

noxilixon, 2023-02-21:
When I reinstalled the requirements I got `ansible-lint 6.13.1 using ansible 2.14.2` as ansible version. This matches the version defined in the requirements.txt:
```
...
ansible >= 2.13
ansible_merge_vars >= 5.0.0
...
```
But this version seems to use new linting rules. When executing the lint with this version I get the following errors:

```
> make lint
pip3 install -q -r requirements.txt
yamllint -d .config/yaml-lint.yml .
ansible-lint -c .config/ansible-lint.yml
WARNING  Listing 22 violation(s) that are fatal

name[play]: All plays should be named.
play.yml:2

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/add_symlinks.yml:2 Task/Handler: Create rc.d directory

ignore-errors: Use failed_when and specify error conditions instead of using ignore_errors.
roles/cfg_openwrt/tasks/fix_permissions.yml:12 Task/Handler: Set permission for prom-catalog-register script

ignore-errors: Use failed_when and specify error conditions instead of using ignore_errors.
roles/cfg_openwrt/tasks/fix_permissions.yml:18 Task/Handler: Set permission for replace_secrets script

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/imagebuilder.yml:2 Task/Handler: Create images dir

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/imagebuilder.yml:16 Task/Handler: Create Download directory

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/imagebuilder.yml:26 Task/Handler: Download Imagebuilder if upstream is newer

no-changed-when: Commands should not change things if nothing needs doing. (warning)
roles/cfg_openwrt/tasks/imagebuilder.yml:32 Task/Handler: Copy Local Imagebuilder

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/imagebuilder.yml:45 Task/Handler: (Re)-Create workdir

no-changed-when: Commands should not change things if nothing needs doing. (warning)
roles/cfg_openwrt/tasks/imagebuilder.yml:56 Task/Handler: Read instruction set

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/imagebuilder.yml:79 Task/Handler: Add falter feed key

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/imagebuilder.yml:108 Task/Handler: Write build log to output dir

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/imagebuilder.yml:133 Task/Handler: Copy Image to Image dir

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/main.yml:29 Task/Handler: Create directory

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/main.yml:49 Task/Handler: Add poemgr config if poemgr target

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/main.yml:65 Task/Handler: Configure ath10k-ct driver

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/render_etc.yml:20 Task/Handler: Create /etc directories for template output

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/render_etc.yml:30 Task/Handler: Create /etc directories for static files

no-free-form: Avoid using free-form when calling module actions. (template)
roles/cfg_openwrt/tasks/render_etc.yml:40 Task/Handler: Render /etc templates

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/render_etc.yml:40 Task/Handler: Render /etc templates

no-free-form: Avoid using free-form when calling module actions. (copy)
roles/cfg_openwrt/tasks/render_etc.yml:48 Task/Handler: Copy /etc static files

risky-file-permissions: File permissions unset or incorrect.
roles/cfg_openwrt/tasks/render_etc.yml:48 Task/Handler: Copy /etc static files

Read documentation for instructions on how to ignore specific rule violations.

                           Rule Violation Summary                           
 count tag                    profile  rule associated tags                 
     1 name[play]             basic    idiom                                
     2 no-free-form           moderate syntax, risk                         
    15 risky-file-permissions safety   unpredictability                     
     2 ignore-errors          shared   unpredictability                     
     2 no-changed-when        shared   command-shell, idempotency (warning) 

Failed after min profile: 20 failure(s), 2 warning(s) on 767 files.
```

So we should fix these warnings or change the linting configs I think.

Noki, 2023-02-21:
I would try to fix all of them as the result is otherwise upnpredictable on different systems, and the fix seems quite easy: https://ansible-lint.readthedocs.io/rules/risky-file-permissions/

FFHener, 2023-04-24:
Was fixed in #442

closed 2023-04-24 by noxilixon