freifunk-berlin/bbb-configs #426: OWE Transition Mode

pktpls, 2023-03-07:
In the past couple of months we tried out OWE Transition Mode on the "berlin.freifunk.net" SSID. This has been rolled back because of a bug:

pktpls, 2023-06-02:
In the meantime @PolynomialDivision has updated OpenWrt hostapd (openwrt/openwrt#12178), but it looks like the EAPOL-key bug persists.

Which makes me think that one of Openwrt's patches to hostapd is responsible. Primary candidates would be the cryptography-related and wifi-related patches, I guess: https://github.com/openwrt/openwrt/tree/main/package/network/services/hostapd/patches

pktpls, 2023-07-14:
It works with OpenSSL :+1: 

With WolfSSL it's broken :-1: some connection attempts with the aforementioned `EAPOL-Key` error from hostapd.

With Mbed TLS it's broken too :-1: all connection attempts fail, but without any error message from hostapd.

PolynomialDivision, 2023-07-15:
I think in comparison openssl is taking way more flash? I think that's why we currently stay with wolfssl or mbedtls?

pktpls, 2023-09-08:
Not reproducible with WolfSSL on current snapshot mt76 / zyxel_nwa50ax, I'll also have to try that Fritzbox again

pktpls, 2023-09-08:
Bug still triggers with WolfSSL and current OpenWrt snapshot on ipq40xx / avm_fritzbox-4040

pktpls, 2023-10-20:
Update:

The issues with WolfSSL and MbedTLS are specific to ipq40xx and have been reproduced on avm_fritzbox-4040 and mikrotik_wap-ac.

We can easily work around it by installing OpenSSL in our ipq40xx target file.

pktpls, 2023-10-20:
Another fixable OWE issue:

The Broadcom `bcmwl` driver on Linux can't deal with OpenWrt's default `wpa_key_mgmt` settings: https://bugs.launchpad.net/ubuntu/+source/bcmwl/+bug/1956162 - I reproduced this with my friend's old Fedora Macbook Air.

We should test setting `wpa_key_mgmt=WPA-PSK SAE`.

pktpls, 2024-04-24:
iOS now does OWE but only with devices 2021++ https://support.apple.com/en-gb/guide/deployment/dep3b0448c58/web

> iOS 16, iPadOS 16.1 and macOS 13, or later, support a way to secure open Wi-Fi networks, using the standard Opportunistic Wireless Encryption. This standard is supported on the following devices: