freifunk-berlin/bbb-configs #429: Display of contact details is missing in Luci

FFHener, 2023-03-14:
If you go to the [contact-tab](https://habersaath-core.olsr/cgi-bin/luci/freifunk/contact) in the GUI of the core-router the contact details are missing although they are set:
```
root@habersaath-core:~# cat /etc/config/freifunk 
config public 'contact'
	list contact 'noc@stadtfunk.net'
	option location 'Leerstand Hab-ich-saath'

config public 'community'
	option homepage 'http://freifunk.net'
	option name 'berlin'
```
This might be related to #320. I saw this issue so far only at locations that dont use the communityflag.

Akira25, 2023-03-14:
Thank you for pointing this out. This is defnitely related to #320. As the LuCI-App looks for a specific configuration file format, which is violated by #320, it can't find the correspondent field. I'm not sure, if this is a real issue though, as per my understanding the bbb-config nodes get configured by ansible only.

On the map, the contact details get displayed correctly:
[image]

If we will fix this issue, it would need to be done in the freifunk-mod package from falter-packages-repo.

FFHener, 2023-04-11:
@Akira25 Feel free to move the issue there. I dont have the rights for it.

Akira25, 2023-04-11:
Are you sure, this is a real issue? As far As I understand, LuCI shouldn't be used on bbb-config nodes anyway. So I suppose, that this issue shouldn't be a real issue at all, right?

FFHener, 2023-04-13:
I think it would be nice to have it because it helps other people to contact the maintainer of a location if they want to mesh with it. Of course you could find it in hopglass, but i think it is a more clean approach it is working there as well on all nodes regardless which method is used to configure them.