freifunk-berlin/bbb-configs #438: Sometimes no OLSR-Routing after tunnelmanager rebuilds tunnel

FFHener, 2023-03-27:
The problem is happening among other nodes at habersaath-core. A 5G-Antenna is used as an uplink for Wireguard. After (properly) the Internetconnection is lost tunnelmanager builds up a new tunnel, Babel for IPv6 is running again, but OLSR is having some trouble. This might be related to [tunnelmanager](https://github.com/freifunk-berlin/falter-packages/tree/master/packages/falter-berlin-tunnelmanager) adding Interfaces in OLSR without restarting it. Running ``/etc/init.d/olsrd restart`` is fixing the problem temporarly. 

The Setup at Habersaathstr. was running flawlessly for month before, but suddenly had problems almost every day.

Noki, 2023-04-04:
I think we should move this to the falter-packages repo, as tunnelmanager lives there: https://github.com/freifunk-berlin/falter-packages

FFHener, 2023-04-10:
Sounds good, i dont have the rights to move it.