freifunk-berlin/bbb-configs #493: Some Websites not reachable

FFHener, 2023-06-23:
Since some time i found myself having trouble reaching certain websites. My first assumption was IPv4-Blocking at the gateways, but I got different problems (tcp-reset <> ssl-error) at different gateways. I had this symptoms:

* Computer: Some Websites have a timeout with Chrome and Firefox
* Phone: 
  * without VPN: some Websites have a timeout with Chrome
  * with VPN: Websites are reachable
* Router: Curl could get the content from the website

I had similar issues with duckduckgo and bitwarden at different bbb-config-nodes, but didnt had time then to proper document it. I heard that other had simular problems with at least Bitwarden. Here are some findings from today and 2 days ago:

Duckduckgo
```
> mtr -r duckduckgo.com
Start: 2023-06-23T02:15:19+0200
HOST:                              Loss%   Snt   Last   Avg  Best  Wrst StDev
[...]
  2.|-- mid11.ak36-gw.olsr         0.0%    10   38.2  36.5  34.1  40.4   1.8
  3.|-- a36s-cr1.berlin.freifunk.  0.0%    10   40.5  37.5  34.0  44.0   3.0
  4.|-- retn.a36.community-ix.de   0.0%    10   37.8  41.1  35.1  49.9   4.6
  5.|-- ae1-5.RT.IRX.FKT.DE.retn.  0.0%    10   46.8  45.6  41.4  48.6   2.5
  6.|-- retn.ier01.fra32.ntwk.msn  0.0%    10   45.7  45.3  43.2  47.7   1.4
  7.|-- ae31-0.icr01.fra21.ntwk.m  0.0%    10   46.1  48.3  44.5  54.6   3.2
  8.|-- be-120-0.ibr02.fra21.ntwk 20.0%    10   48.3  50.5  47.6  53.6   1.9
  9.|-- be-8-0.ibr02.ams30.ntwk.m  0.0%    10   48.8  49.5  47.3  52.8   1.7
 10.|-- ae120-0.icr01.ams30.ntwk.  0.0%    10   51.3  50.3  47.3  55.3   2.3
 11.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 12.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 13.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 14.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 15.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 16.|-- 40.114.177.156             0.0%    10   52.9  51.6  46.8  56.5   3.1
 
> curl -v duckduckgo.com
*   Trying 40.114.177.156:80...
* Connected to duckduckgo.com (40.114.177.156) port 80 (#0)
> GET / HTTP/1.1
> Host: duckduckgo.com
> User-Agent: curl/7.74.0
> Accept: */*
> 
* Recv failure: Connection reset by peer
* Closing connection 0
curl: (56) Recv failure: Connection reset by peer
```

```
> mtr -r duckduckgo.com
Start: 2023-06-21T09:38:39+0200
HOST:                               Loss%   Snt   Last   Avg  Best  Wrst StDev
[...]
  2.|-- mid10.saarbruecker-gw.ols  0.0%    10   17.4  15.6  12.9  17.7   1.7
  3.|-- ge-0-1-8-0.bsa1-cpe1.de.s  0.0%    10   14.2  17.9  14.2  31.6   5.0
  4.|-- xe-1-0-4-0.blu1-r2.de.sys  0.0%    10   18.1  16.6  14.9  18.2   1.4
  5.|-- ae3-0.bki1-r2.de.syseleve  0.0%    10   18.0  19.5  14.5  35.1   5.9
  6.|-- microsoft2.bcix.de         0.0%    10   14.6  16.5  14.3  19.6   2.0
  7.|-- ae22-0.icr02.ber20.ntwk.m  0.0%    10   19.0  17.3  13.6  20.9   2.4
  8.|-- be-102-0.ibr01.ber20.ntwk 20.0%    10   27.1  29.6  25.8  34.8   4.0
  9.|-- be-5-0.ibr01.ham31.ntwk.m  0.0%    10   26.4  29.3  26.2  33.0   2.3
 10.|-- be-3-0.ibr01.ham30.ntwk.m 30.0%    10   26.3  29.0  25.5  32.2   2.7
 11.|-- be-12-0.ibr05.ams06.ntwk.  0.0%    10   27.7  29.1  26.4  31.0   1.5
 12.|-- ae142-0.icr04.ams06.ntwk.  0.0%    10   26.9  27.5  25.2  35.3   3.0
 13.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 14.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 15.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 16.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 17.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
 18.|-- 40.114.177.156             0.0%    10   30.0  29.3  26.0  32.2   2.2
 
 > curl -v duckduckgo.com
*   Trying 40.114.177.156:80...
* Connected to duckduckgo.com (40.114.177.156) port 80 (#0)
> GET / HTTP/1.1
> Host: duckduckgo.com
> User-Agent: curl/7.74.0
> Accept: */*
> 
* Recv failure: Connection reset by peer
* Closing connection 0
curl: (56) Recv failure: Connection reset by peer
```
Bitwarden
```
> mtr -r bitwarden.com
Start: 2023-06-23T02:19:26+0200
HOST:                             Loss%   Snt   Last   Avg  Best  Wrst StDev
[...]
  2.|-- mid11.ak36-gw.olsr         0.0%    10   36.0  35.1  32.9  37.2   1.2
  3.|-- a36s-cr1.berlin.freifunk.  0.0%    10   40.4  36.9  32.9  40.4   2.3
  4.|-- retn.a36.community-ix.de   0.0%    10   43.6  37.3  34.8  43.6   2.7
  5.|-- ae1-5.RT.IRX.FKT.DE.retn.  0.0%    10   45.7  45.3  40.4  60.0   5.8
  6.|-- GW-Fastly.retn.net         0.0%    10   44.7  45.3  41.4  50.5   2.6
  7.|-- 199.232.194.22             0.0%    10   42.1  45.2  42.1  51.6   2.6
  
> curl -v https://bitwarden.com
*   Trying 199.232.194.22:443...
* Connected to bitwarden.com (199.232.194.22) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* OpenSSL SSL_connect: Connection reset by peer in connection to bitwarden.com:443 
* Closing connection 0
curl: (35) OpenSSL SSL_connect: Connection reset by peer in connection to bitwarden.com:443 
```

```
> mtr -r bitwarden.com
Start: 2023-06-21T09:39:17+0200
HOST:                               Loss%   Snt   Last   Avg  Best  Wrst StDev
[...]
  2.|-- mid10.saarbruecker-gw.ols  0.0%    10   16.6  17.0  14.6  20.9   1.9
  3.|-- ge-0-1-8-0.bsa1-cpe1.de.s  0.0%    10   18.2  18.7  14.4  34.3   5.8
  4.|-- ktel.bak1-r1.syseleven.ne  0.0%    10   17.1  18.6  15.7  24.9   3.0
  5.|-- ae0-0.blu1-r2.de.syseleve  0.0%    10   26.0  26.2  22.8  32.9   3.1
  6.|-- ae1.fix5-r1.de.syseleven.  0.0%    10   34.6  28.4  24.1  34.6   3.4
  7.|-- ???                       100.0    10    0.0   0.0   0.0   0.0   0.0
  8.|-- 199.232.194.22             0.0%    10   24.7  26.1  23.3  32.1   2.8

> curl -v https://bitwarden.com
*   Trying 199.232.194.22:443...
* Connected to bitwarden.com (199.232.194.22) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* OpenSSL SSL_connect: Connection reset by peer in connection to bitwarden.com:443 
* Closing connection 0
curl: (35) OpenSSL SSL_connect: Connection reset by peer in connection to bitwarden.com:443 
```

Stackoverflow
```
> mtr -r stackoverflow.com
Start: 2023-06-23T02:35:17+0200
HOST:                               Loss%   Snt   Last   Avg  Best  Wrst StDev
[...]
  2.|-- mid11.ak36-gw.olsr         0.0%    10   34.2  36.2  33.8  39.2   1.8
  3.|-- a36s-cr1.berlin.freifunk.  0.0%    10   36.1  36.0  34.3  39.3   1.7
  4.|-- retn.a36.community-ix.de   0.0%    10   35.9  36.5  34.8  38.0   1.0
  5.|-- ae1-5.RT.IRX.FKT.DE.retn.  0.0%    10   41.3  44.0  41.3  47.1   2.0
  6.|-- GW-Fastly.retn.net         0.0%    10   41.7  44.1  41.5  50.8   3.5
  7.|-- 151.101.129.69             0.0%    10   41.2  44.8  41.2  50.7   2.5

> curl -v https://stackoverflow.com
*   Trying 151.101.193.69:443...
* Connected to stackoverflow.com (151.101.193.69) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* OpenSSL SSL_connect: Connection reset by peer in connection to stackoverflow.com:443 
* Closing connection 0
curl: (35) OpenSSL SSL_connect: Connection reset by peer in connection to stackoverflow.com:443 
```

FFHener, 2023-06-27:
A friend of mine reported simular Issues with a Falter-Router (Archer C7 v5.0, Firmware 1.2.3) with tunneldigger. The IP that was used is pointing to netcup, so i think he was connected to ``f.tunnel``. He reported:
* Cannot connect to the DNS from dns.adguard.com
* Apple Update Service from iPhones didn't worked
* Aurorastore don't work
** The Website from this project showed a blocking page from Cloudflare

FFHener, 2023-07-02:
Just wanted to add that ``strato.de`` is another site with troubles.

FFHener, 2023-07-08:
A major part of this errors got resolved with #498. However, the problems with the tunnelmanagerserver are likely still there, because this change didnt affect them.

pktpls, 2023-10-01:
ohlauer-gw IPv4 blocked by Duckduckgo

pktpls, 2023-10-10:
The SysEleven IPv4 block 176.74.56.0/21 is part of a blocklist: https://iplists.firehol.org/?ipset=firehol_webserver -- we use 176.74.57.43/31 at Saarbruecker and 176.74.57.19/31 at Ohlauer

pktpls, 2023-10-10:
Parts of our Förderverein 77.87.48.0/21 block (AK36, L105, Strom) are in lists in that repo as well: https://github.com/firehol/blocklist-ipsets

pktpls, 2023-11-25:
The problem was found: faulty MTU settings

closed 2023-11-25 by pktpls