freifunk-berlin/bbb-configs #512: "daemon.warn odhcpd: No default route present, overriding ra_lifetime!"

FFHener, 2023-07-21:
After flashing a new router for #511 this warn-message randomly appeared and dissapered again 2 times. I'm unshure where the root of this Error is so here are a bunch of Infos hopefully being helpful. The timeline was: I installed the router in the afternoon, it suddenly stopped working during evening, but was still reachable via IPv6-link local. During the night it suddenly worked again (according to monitoring), but I found similar error-message during a timespan of 10 min in the morning.

Initial logread and routes during the evening:
```
~# logread
daemon.warn odhcpd[1519]: No default route present, overriding ra_lifetime!

~# ip r
10.31.137.48/28 dev switch0.42 proto kernel scope link src 10.31.137.49 
10.31.211.64/26 dev switch0.40 proto kernel scope link src 10.31.211.65 

~# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1): 56 data bytes
ping: sendto: Network unreachable

~# ip r s t all
10.31.137.48/28 dev switch0.42 proto kernel scope link src 10.31.137.49 
10.31.211.64/26 dev switch0.40 proto kernel scope link src 10.31.211.65 
local 10.31.137.49 dev switch0.42 table local proto kernel scope host src 10.31.137.49 
broadcast 10.31.137.63 dev switch0.42 table local proto kernel scope link src 10.31.137.49 
local 10.31.174.8 dev switch0.10 table local proto kernel scope host src 10.31.174.8 
local 10.31.174.9 dev switch0.11 table local proto kernel scope host src 10.31.174.9 
local 10.31.174.10 dev switch0.20 table local proto kernel scope host src 10.31.174.10 
local 10.31.211.65 dev switch0.40 table local proto kernel scope host src 10.31.211.65 
broadcast 10.31.211.127 dev switch0.40 table local proto kernel scope link src 10.31.211.65 
local 127.0.0.0/8 dev lo table local proto kernel scope host src 127.0.0.1 
local 127.0.0.1 dev lo table local proto kernel scope host src 127.0.0.1 
broadcast 127.255.255.255 dev lo table local proto kernel scope link src 127.0.0.1 
default from 2001:bf7::/32 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:1200::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:2a00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:2e00::/60 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:3200::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:3300::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:3400::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:3700::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:3f00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:4000::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:4100::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:4a00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:4b00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:5400::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:750:5600::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:760:1000::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:760:2200::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:760:2600::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:760:2700::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:760:2a00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:760:2b00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:760:2c00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:790::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:800:f00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:810:1000::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:300::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:800::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:e00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:1000::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:1500::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:1800::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:2300::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:2400::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:2500::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:820:2600::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:830:b00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:830:1f00::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:830:2600::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:830:3000::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:830:6700::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:830:8000::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
2001:bf7:830:8200::/56 via fe80::1411:58ff:fe2b:4cff dev switch0.10 proto babel metric 1024 onlink pref medium
[... 602 lines cut ...]
	  Advertised networks (total 2)
	    10.230.36.28/30, 10.31.174.96/28
	TC-LQ Message (0xca), originator 10.31.20.141, ttl 2, hop 6
[...]
```
Logread from next morning:
```
Thu Jul 20 07:15:00 2023 cron.err crond[3335]: USER root pid 7292 cmd killall -HUP dnsmasq
Thu Jul 20 07:15:00 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Jul 20 07:15:00 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 10 names
Thu Jul 20 07:15:00 2023 daemon.info dnsmasq[1]: read /tmp/hosts/olsr - 1373 names
Thu Jul 20 07:15:00 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Jul 20 07:15:01 2023 daemon.err collectd[2147]: network plugin: getaddrinfo (monitor.berlin.freifunk.net, (null)) failed: Try again
Thu Jul 20 07:15:18 2023 daemon.warn odhcpd[1519]: No default route present, overriding ra_lifetime!
Thu Jul 20 07:15:18 2023 daemon.warn odhcpd[1519]: No default route present, overriding ra_lifetime!
Thu Jul 20 07:16:06 2023 daemon.err collectd[2147]: network plugin: getaddrinfo (monitor.berlin.freifunk.net, (null)) failed: Try again
Thu Jul 20 07:17:00 2023 cron.err crond[3335]: USER root pid 7293 cmd test -e /usr/sbin/owm.sh && /usr/sbin/owm.sh
Thu Jul 20 07:17:28 2023 daemon.err collectd[2147]: network plugin: Successfully resolved "monitor.berlin.freifunk.net".
Thu Jul 20 07:20:00 2023 cron.err crond[3335]: USER root pid 7387 cmd killall -HUP dnsmasq
Thu Jul 20 07:20:00 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Jul 20 07:20:00 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 10 names
Thu Jul 20 07:20:00 2023 daemon.info dnsmasq[1]: read /tmp/hosts/olsr - 1373 names
Thu Jul 20 07:20:00 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Jul 20 07:25:00 2023 cron.err crond[3335]: USER root pid 7396 cmd killall -HUP dnsmasq
```

FFHener, 2023-11-28:
The issue occured again, it looks like a olsr-problem:

```
root@liese-21-core:~# wget -O - http://127.0.0.1:2006 > olsr.txt

root@liese-21-core:~# cat olsr.txt 
Table: Neighbors
IP address	SYM	MPR	MPRS	Will.	2-hop count
10.31.83.60	YES	YES	NO	3	1

Table: Links
Local IP	Remote IP	Hyst.	LQ	NLQ	Cost
10.31.174.8	10.31.83.61	0.000	0.016	1.000	INFINITE

Table: Routes
Destination	Gateway IP	Metric	ETX	Interface

Table: HNA
Destination	Gateway
10.31.137.48/28	10.31.174.8
10.31.211.64/26	10.31.174.8
10.31.127.96/27	10.31.124.199
10.31.81.0/27	10.31.79.176
10.31.81.128/25	10.31.79.176
10.36.102.0/25	10.36.102.129
10.36.102.128/30	10.36.102.129
10.36.102.132/30	10.36.102.129
10.36.102.136/30	10.36.102.129
0.0.0.0/0	10.36.102.129
10.31.228.192/26	10.31.228.160
10.31.228.128/27	10.31.228.160
10.31.108.16/28	10.31.104.220
10.31.183.192/26	10.31.183.145
10.31.183.160/27	10.31.183.145
10.31.183.128/28	10.31.183.145
10.31.173.160/27	10.31.173.144
10.31.173.128/28	10.31.173.144
10.36.69.96/27	10.31.28.17
10.36.160.56/30	10.31.28.17
10.31.130.128/27	10.31.130.160
0.0.0.0/0	10.31.130.160
10.31.161.208/28	10.31.161.168
10.31.78.240/28	10.31.78.131
10.31.85.208/28	10.31.42.89
10.230.205.8/29	10.36.254.82
10.36.90.224/27	10.31.29.65
10.31.83.48/29	10.31.83.56
10.230.106.16/28	10.31.14.203
10.31.127.80/28	10.31.124.205
10.31.83.112/28	10.31.83.60
10.31.83.192/26	10.31.83.60
10.230.104.0/28	10.31.14.148
10.31.199.0/26	10.31.199.254
10.230.162.144/28	10.31.19.208
10.36.243.160/30	10.36.243.192
10.36.243.164/30	10.36.243.192
10.36.243.168/30	10.36.243.192
10.36.243.172/30	10.36.243.192
10.36.243.176/30	10.36.243.192
10.36.243.180/30	10.36.243.192
10.36.243.184/30	10.36.243.192
10.36.244.0/24	10.36.243.192
10.36.243.128/28	10.36.243.192
10.230.108.96/28	10.31.14.154

Table: MID
IP address	(Alias)+
10.31.27.245		10.36.193.101
10.31.104.220		10.31.104.221
10.31.36.45		10.36.197.233
[... 343 lines cut ...]
10.31.42.112	10.31.77.254	0.940	1.000	INFINITE
10.31.1.128	10.31.77.254	1.000	1.000	INFINITE
10.31.18.254	10.31.77.254	1.000	0.917	INFINITE
10.30.96.54	10.30.96.254	1.000	1.000	INFINITE
10.31.199.253	10.31.199.254	0.886	0.983	1.146

Table: Gateways
Status Gateway IP                                    ETX             Hopcnt Uplink    Downlnk   IPv4    IPv6 Prefix

Table: Interfaces
Name	State	MTU	WLAN	Src-Adress	Mask	Dst-Adress
switch0.10	UP	1472	No	10.31.174.8	255.255.255.255	255.255.255.255
switch0.11	UP	1472	No	10.31.174.9	255.255.255.255	255.255.255.255
switch0.20	UP	1472	No	10.31.174.10	255.255.255.255	255.255.255.255

Table: Neighbors
IP address	SYM	MPR	MPRS	Will.	(2-hop address)+
10.31.83.60	YES	YES	NO	3	10.31.79.177

Table: Smart Gateway IPv4
 Originator      Prefix                          Uplink    Downlink  PathCost   IPv4 IPv4-NAT IPv6 Tunnel-Name     Destination     Cost

Version: olsr.org - pre-0.9.9-git_0000000-hash_912fe6e079f7b7b9fd547d8a856e846c

```

FFHener, 2023-12-05:
As the problem accured again the other side magda reported the following in dmesg:

```
[8053815.566598] neighbour: ndisc_cache: neighbor table overflow!
[8053815.566737] neighbour: ndisc_cache: neighbor table overflow!
[8053816.607283] neighbour: ndisc_cache: neighbor table overflow!
[8053816.607458] neighbour: ndisc_cache: neighbor table overflow!
[8053816.928190] neighbour: ndisc_cache: neighbor table overflow!
[8053817.326025] neighbour: ndisc_cache: neighbor table overflow!
```

pktpls, 2023-12-05:
A symptom we saw together with `ndisc_cache: neighbor table overflow` is OLSR link quality oscillating between crap and good repeatedly within a minute.

FFHener, 2024-02-18:
In the end the corerouter didnt received enough energy which caused a lot of weird behaviours. As this is solved im closing this issue for now

closed 2024-02-18 by FFHener