freifunk-berlin/bbb-configs #568: perleberger36-core: Please fix the management adresses

pmelange, 2023-11-07:
gitblame says it was @PolynomialDivision who set it up with all the assignments set to 2. 

https://github.com/freifunk-berlin/bbb-configs/blob/e83ff9f769ad4475e83d892b1eec35fddf648f72/group_vars/location_perleberger36/networks.yml#L92-L103

Noki, 2023-11-08:
#569 fixes most of this. however two of the antennas are offline and the IPs for those remain unclear.

pmelange, 2023-11-08:
https://wiki.freifunk.net/Berlin:Standorte:Heilige-Geist-Kirche_Moabit should have the correct info for any offline devices

pmelange, 2023-11-08:
> https://wiki.freifunk.net/Berlin:Standorte:Heilige-Geist-Kirche_Moabit should have the correct info for any offline devices

I leave change requests.  The two "offline" devices never existed.

pmelange, 2023-11-13:
fix via #569

closed 2023-11-13 by pmelange