freifunk-berlin/bbb-configs #595: Update everything to 23.05

pktpls, 2023-11-25:
The following models are still on 22.03

- glinet_gl-b1300
- mikrotik_routerboard-750gr3
- mikrotik_routerboard-760igs
- ubnt_nanostation-ac-loco
- ubnt_nanostation_loco_m5_xm
- ubnt_nanostation_m2_xm
- ubnt_nanostation_m5_xm
- ubnt_nanostation_m5_xw
- ubnt_rocket-5ac-lite
- ubnt_unifi-6-lite
- ubnt_unifiac-lite
- ubnt_unifiac-mesh
- ubnt_unifiac-pro

pktpls, 2023-12-05:
22.03 will reach End-of-Life in April

Petrella, 2023-12-22:
Isn't Ubiquiti nanostation m5 xm still on 22.03 officially?

pktpls, 2023-12-24:
Yes, all the XM devices only have 32 MB RAM. We'll have to find tricks for that

Petrella, 2023-12-25:
Will that be a problem? In the model files for devices like `model_ubnt_nanostation_m5xm` the `low_mem` option is set to true. So if you build an image luci and other things will not be present to make the image small. Do you have other concerns?

pktpls, 2024-04-05:
22.03 EOL is Thursday next week

pktpls, 2024-07-23:
The Unifi AC devices hds-ffraum and zwingli-dome-2 are on 23.05 now - the procedure is very simple: #909 

The dragonkiez and kotti devices can be upgraded the same way.

pktpls, 2024-07-24:
The Nanostations upgrade fine - they only lose their overlay data, which is a very minor issue for us.

We're ready to remove 22.03

pktpls, 2024-08-03:
We're done, 22.03 is already removed from main

closed 2024-08-03 by pktpls