freifunk-berlin/bbb-configs #696: Correction of the location coordinates in the wiki

xa-y-ax, 2024-01-04:
Please correct the coordinates of the bbb-configs locations in the wiki so that the Line-of-Sight Visualizer can read them correctly.

FFHener, 2024-01-04:
The problem is accuring with the current .kml file. You can see it here:[image]

Next steps:
- [ ] Does the wiki attributes have the correct xoordinates
- [ ] Does util pull the correct coordinates as seen in the [php-file](https://util.berlin.freifunk.net/line-of-sight.php)

FFHener, 2024-01-09:
I found a way to "solve" the problem. The proplem occured because the locations in the wiki didnt had the `Hat Koordinaten`-Attribute. This behavior happens when we push the coordinates via the API. Apparently it doesnt redo the templates after it. 
What "fixed" it for me was to check out the [Fehlerhafte Annotationen](https://wiki.freifunk.net/Attribut:Hat_Koordinaten), klick edit in the article and directly save it without changing anything or giving a edit message.

This way the "edit" doesnt show up in the version history, but the attribute gets correctly renderd.

Steps to reproduce:

0. Have an article without coordinates
1. Push coordinates via API e.g. use bbb-configs
=> Error, the attribute doesnt exist
2. Press `Quelltext bearbeiten`
3. Press `Änderung speichern`
=> Error fixed

FFHener, 2024-01-09:
Wont fix it any further as there is already a migration path towards having the location pages as a subpage of our website

closed 2024-01-09 by FFHener