freifunk-berlin/bbb-configs #751: RFC: Changes to the configuration format to further standardize it

Noki, 2024-01-30:
I talked with @FFHener about ways to further standardize the configuration format. We came up with the following ideas:

- Add model files for non OpenWRT devices (ubiquiti antennas, switches, webcams, ...)
- Add all non OpenWRT devices as hosts alongside the OpenWRT devices
- Add configurations values to individual non OpenWRT hosts (e.g. height+altitude, alignment, ssid, ...)

Doing so will allow us to move some parts of the configuration into the new model files:

- SNMP profiles
- airOS DFS reset (default configuration, with configuration overrides for each host)

In addition this will allow us to:

- derive a proper documentation for each location that could be used to update the wiki / website
- generate configuration files for non OpenWRT devices when building the location

Here are examples of how a model file for a non OpenWRT device could look like:

```
brand_nice: Ubiquiti
model_nice: Wave Long-Range

wireless_devices:
  - name: 11ad_standard
    band: 60g
  - name: 11a_standard
    band: 5g
    htmode_prefix: HE

snmp_profile: af60
```

```
brand_nice: Ubiquiti
model_nice: Nanostation 5AC

wireless_devices:
  - name: 11a_standard
    band: 5g
    htmode_prefix: HE

snmp_profile: airos_8

airos_dfs_reset:
  - username: "ubnt"
    password: "file:/root/pwd"
    daytime_limit: "2-7"
```

The host configuration for such an device could look like this:

```
hosts:
  - hostname: q216-tierpark
    role: ptp-ap
    ptp: true
    model: "ubnt_nanobeam_5ac-gen2"
    height: 15
    altitude: 54
    orientation: ost
    ssid: "q216-tierpark" #to override default ssid (freifunk-{location}-{orientation}-{frequency} / freifunk-{location}-{orientation})
    channel: 128
    channel-width: 40
    username: "root" # to override the default ("ubnt")
    password: "file:/root/tierpark-pwd.txt" # to override the default ("file:/root/pwd.txt)
    daytime_limit: "4-5"  # to override the default ("2-7")
```

We want to discuss this at the community day. Please let us know about any thoughs you have on this.

FFHener, 2024-01-30:
I added some comments in the proposed code. Did you check with the config files from the properitery devices that we need the wireless_devices in this way?

Noki, 2024-01-30:
Well, we need some way to know the band and we also need to know if it is a dual band device, so probably a good idea to just keep the syntax we already have and build on that.

pmelange, 2024-01-30:
It would be cool if a config file for the ubiquiti devices were also generated.  There isn't much to the ubnt config files and on a new device (or after reset) on the first landing page it's possible to upload a config file.

Noki, 2024-01-30:
> It would be cool if a config file for the ubiquiti devices were also generated. There isn't much to the ubnt config files and on a new device (or after reset) on the first landing page it's possible to upload a config file.

thats the plan... ;-)

FFHener, 2024-01-30:
Another aspect would be to move the gateway config over to our default format. I think it should be manageable for everything that is not gateway specific as we allready have a way of describing mesh, management and uplink interfaces.
At first glance it looked like we would need to modify 8 templates to make this work.