freifunk-berlin/bbb-configs #766: collectd: disable logging for locations with AirOS 6 snmp_profiles

FFHener, 2024-02-07:
When a AirOS 6 AP has no Stations connected a snmp-call asking for their names results in a `NoSuchName` error for each parameter. That means a lot of log spamming. As we only need collectd logging when we debug it i would suggest adding a script at [rc.local](https://github.com/freifunk-berlin/bbb-configs/blob/master/roles/cfg_openwrt/templates/common/rc.local.j2) [deactivating logging](https://github.com/openwrt/packages/blob/75b72c402f714af08c18a635926b3210622de818/utils/collectd/files/collectd.init#L599) after boot.

Noki, 2024-02-07:
> As we only need collectd logging when we debug it

I disagree. We want to have historical data for our sites so we can identify performance degradations. We have had antennas whose alignment has deteriorated over time, e.g. due to wind, snow load or unstable mounting, and we have had antennas that have deteriorated due to water ingress. We can only detect all of this with continuous monitoring.

FFHener, 2024-02-07:
Thats a misunderstabding. Collectd should still log the snmp-data, but not the errors to logd locally on the device

Noki, 2024-02-07:
Got it! Yes, that's what we want.

commit ae3bc7e1 (2024-02-10): templates: deactivate stderr of collectd if it monitors airos_6

When collectd wants to get the Names of Stations via snmp connnected
to a device running AirOS 6 that have none connected it reports a
"NoSuchName" error to logd resulting in a lots of logspamming. At
chris-core it was 60 per minute.

This Script will change the option "procd_set_param stderr  1" to "0"
in the file /etc/init.d/collectd to deactivate logging of all errors.
The monitoring will still work. I prefer not having logs over logspamming
as the logfiles will get useless otherwise. It is still possible to
change this option [... cut]
PR #769 (2024-02-10): templates: deactivate stderr of collectd if it monitors airos_6
When collectd wants to get the Names of Stations via snmp connnected to a device running AirOS 6 that have none connected it reports a "NoSuchName" error to logd resulting in a lots of logspamming. At chris-core it was 60 per minute.

This Script will change the option "procd_set_param stderr  1" to "0" in the file /etc/init.d/collectd to deactivate logging of all errors. The monitoring will still work. I prefer not having logs over logspamming as the logfiles will get useless otherwise. It is still possible to change this option  [... cut]
commit f955453a (2024-02-10): templates: deactivate stderr of collectd if it monitors airos_6

When collectd wants to get the Names of Stations via snmp connnected
to a device running AirOS 6 that have none connected it reports a
"NoSuchName" error to logd resulting in a lots of logspamming. At
chris-core it was 60 per minute.

This Script will change the option "procd_set_param stderr  1" to "0"
in the file /etc/init.d/collectd to deactivate logging of all errors.
The monitoring will still work. I prefer not having logs over logspamming
as the logfiles will get useless otherwise. It is still possible to
change this option [... cut]
closed 2024-02-10 by FFHener