freifunk-berlin/bbb-configs #900: multiple: missing mac_overrides of mikrotik sxtsq devices

Noki, 2024-07-19:
We have multiple bbb-configs locations that miss macaddress overrides for mikrotik_sxtsq devices. This should be fixed.

The mac_address of the device can be retrieved using `cat /sys/firmware/mikrotik/hard_config/mac_base`

- [x] Identify all locations with missing mac_overrides
- [ ] add missing mac_overrides
  - [x] mlk-nk (Noki)
  - [x] spitta13 (Noki)
  - [x] linie206 (@FFHener)
  - [x] stadalbert (@FFHener)
  - [x] habersaath (Noki)
  - [ ] gub37
  - [ ] fesev (@FFHener )
  - [ ] mahalle
  - [x] q216 (@FFHener )
  - [x] zwingli (Noki) (1/2 due to missing password see https://github.com/freifunk-berlin/meta/issues/43)
  - [x] teufelssecurity
- [ ] try to develop some kind of linter action to prevent this from happening in the future

Please mark the location with your name if you will fix it.

Noki, 2024-07-19:
> Where are mac overrides missing at fesev?

Looks like I did a mistake there...

ZuquetuPlexis, 2024-09-21:
PR: https://github.com/freifunk-berlin/bbb-configs/pull/973
- [X] temp fix
- [ ] hacrafu-fiocchi-core
- [ ] hacrafu-fiocchi-core
- [ ] hacrafu-fiocchi-core

Noki, 2024-09-22:
I adjusted the check to only run on changed files.