freifunk-berlin/bbb-configs #935: RFC: increase default value of DHCP leasetime

Noki, 2024-08-10:
We currently set a DHCP leasetime of only 5 minutes: 
https://github.com/freifunk-berlin/bbb-configs/blob/cad7136daf4a9bfb4719292aabb147721e0fbabe/roles/cfg_openwrt/templates/corerouter/config/dhcp.j2#L41

The implication of this is that DHCP leases need to be renewed quite frequently. Renewing a DHCP lease requires management traffic that results in WLANs using basic data rates, which uses more RF time than normal transmit data rates.

- Should we increase the default DHCP leasetime to reduce RF time?
- If so, what would you consider a good default value?
- Should we make DHCP leasetime a config option so it can be overwritten for specific locations?

pktpls, 2024-08-10:
> Renewing a DHCP lease requires management traffic

I dont understand - DHCP is just regular UDP broadcast, and for renewal it's usually even unicast. Are you seeing performance problems?

I think it's fine to make it configurable. The other side of the trade-off is that locations with lots of clients need to clean up unused IPs quickly.

pmelange, 2024-08-15:
I believe the reason for the 5 minute lease is because of roaming between ff networks. If a device losses it's connection to node A and reconnects to node B, it will not request a new lease because the old one is still good. If the timeline is large (1h) then the user will have a "bad experience". But with a 5 min lease the chances that someone will even notice is reduced a lot.

I think 5 minutes is best