freifunk-berlin/bbb-configs #969: Document and enforce mesh_metric

spolack, 2024-09-17:
We want to start routing IPv4 via Babel soonish. Lets make sure that the metrics are in the desired state to have a deterministic routing experience :)

- [ ] Agree on metrics concept
- [ ] Document metrics
- [ ] Ensure metrics are set over most criticial PoPs

spolack, 2024-09-17:
RFC 0.1 
| Metric | \>=Speed | Remarks |
| --- | --- | --- |
| 128-511 | &nbsp;5 Gbit/s | For future use |
| 512-4095 | 1 Gbit/s | 60Ghz |
| 4096-8191 | 100 Mbit/s | 5GHz |
| 8192-16383 | 50 Mbit/s | 5GHz PtMP |
| 16384-32767 | 10Mbit/s | 2GHz / Default) |

Metric: The first value should be tried first. In case of suboptimal routing the the range can be used as wiggle-room for individual traffic engineering.

By applying this scheme we would prefer:

- 4 5GBit/s links over single 1Gbit Link
- 8 1Gbit/s links over single 100Mbit/s link
- ...

Noki, 2024-09-19:
I think we should be a bit more granular especially in the area between 1 Gbit/s and 100 Mbit/s so we can set values that match the speed of the connections more closely e.g. by measuring it with iperf3 or looking at the capacities reported by UISP. In my opinion this would help in areas where there are multiple 5 GHz paths with a lot of variance in bandwidth. I know that this was meant with `wiggle room` but I would prefer a more detailed table as it leaves less room for interpretation. In addition I think it is also a good idea to have 2.5 Gbit/s as we already have it with the airFiber 60 Xtreme-Range (ak36<->teufelsberg), eventhough the cable connection/core-router is currently the limit there.

My proposal would be as follows:

| **Metric**      | **Max Speed**   | **Min Speed >**   | **Remarks**                           |
|-----------------|-----------------|-----------------|---------------------------------------|
| 64 - 127        | 10 Gbit/s       | 5 Gbit/s        | Fiber                                 |
| 128 - 255       | 5 Gbit/s        | 2.5 Gbit/s      | Fiber                                 |
| 256 - 511       | 2.5 Gbit/s      | 1 Gbit/s        | 60 GHz wireless/ethernet              |
| 512 - 1023      | 1 Gbit/s        | 500 Mbit/s      | 60 GHz wireless/ethernet              |
| 1024 - 2047     | 500 Mbit/s      | 250 Mbit/s      | 5 GHz wireless                        |
| 2048 - 3071     | 250 Mbit/s      | 100 Mbit/s      | 5 GHz wireless                        |
| 3072 - 4095     | 100 Mbit/s      | 50 Mbit/s       | 5 GHz wireless                        |
| 4096 - 6143     | 50 Mbit/s       | 25 Mbit/s       | 5 GHz / 2 GHz wireless                |
| 6144 - 8191     | 25 Mbit/s       | 10 Mbit/s       | 5 GHz / 2 GHz wireless                |
| 8192 - 12287    | 10 Mbit/s       | 5 Mbit/s        | 5 GHz / 2 GHz wireless                |
| 12288 - 16383   | 5 Mbit/s        | 1 Mbit/s        | 5 GHz / 2 GHz wireless                |
| 16384 - 32767   | 1 Mbit/s        | 0 Mbit/s       | Fallback, management link |

A connection like emma<->rhnk that has a capacity of 273 Mbit/s would be in the metric range of 1024 - 2047, with a default metric of 1024 or with an adjusted metric set to something closer to 2047.

spolack, 2024-09-20:
Thinking of whether it would be nice to find a way to have metrics in a bidirectional way, as usually 5GHz strongly depends on the direction. And how to model the data for PtMP APs per individual Station. Also it would be beneficial to find a way, how to aritificially double the cost for traffic entering and leaving the location on the same interface. (For instance wilgu10-sama -> sama-sued-60ghz -> w38b). In that case the cost of the routes from w38b should be added twice at sama-sued-60ghz, because the possible bandwidth is only the half.

And a third thought is that we could have a helper function, either written as a jinja2 macro or attach a python lookup plugin to our playbooks, where we pass the estimated Bandwidth, instead of manually calculating the cost.
example:
```
mesh_metric: "{{ METRIC_FROM_BW(2000) }}"
```

Noki, 2024-10-31:
I think having a function that translates bandwidth into a mesh_metric is a really good idea. Since most of the traffic is RX we can base everything on measured or estimated RX bandwidth between the node and the neighbor and also apply this to tunspace uplinks.

I can follow your argument with traffic entering and leaving the same location at the same interface and would love to see a solution eventhough I think we are also fine without one. In our current network topology (https://github.com/freifunk-berlin/bbb-configs/pull/1010 - contains a map which shows gateway selection) this kind of traffic is either totally valid (traffic within our network), or a cause of routing when we have outages and can't really avoid it.