freifunk-berlin/bbb-configs #995: Replace babel with bird2

spolack, 2024-10-03:
In order to shrink our zoo of routing daemons i'd propose to replace babeld with bird2. I'm currently running a PoC configuration at wilgu10-core. This also allows us to get rid of the "hack" trying to let bird workaround missing features (kernel export filter) in babeld. Other communities like freifunk franken also switched to bird because it's performing better. Even Jech thinks that birds implemenation is superior.

PoC Config running at wilgu10-core. From what i can tell it seems to work and is interoperable with babeld.

```
root@wilgu10-core:~# cat /etc/bird.conf 
log syslog all;
debug protocols all;

ipv6 sadr table v6_main;

ipv4 table v4_babel_src;
ipv4 table v4_babel_ff;
ipv4 table v4_babel_default;

protocol device {
};

protocol direct {
	interface "lo";
	ipv6 sadr {
		table v6_main;
	};
}

protocol babel {
	randomize router id yes;
        ipv6 sadr {
                table v6_main;
                import where (net.len = 56) || (net = ::/0 from 2001:bf7::/32 );
                export filter {
			if net = 2001:bf7:860:900::/56 from ::/0 then accept;
			if source = RTS_BABEL then accept;
			reject;
		};
        };
        ipv4 {                                          
                table v4_babel_src;                                   
                import all;      
                export where source = RTS_BABEL;
        };
	
	interface "switch0.10" {
		type wireless;
		rxcost 128;
	};
        interface "switch0.11" {
                type wireless;
                rxcost 1024;
        };
        interface "switch0.20" {
                type wireless;
                rxcost 2048;
        };
}

protocol kernel {       
        ipv6 sadr {                  
                table v6_main;
                export all;
        };       
}   

protocol kernel {
        kernel table 12;
	ipv4 {
                table v4_babel_src;
		export all;
	};
}

protocol kernel { 
        kernel table 10;
	ipv4 {
                table v4_babel_ff;
		export all;
	};
}

protocol kernel { 
        kernel table 11;
        ipv4 {             
        	table v4_babel_default;
                export all;
        };                 
}

protocol pipe {
	table v4_babel_src;
	peer table v4_babel_ff;
	export where net != 0.0.0.0/0;
}

protocol pipe {
        table v4_babel_src;
        peer table v4_babel_default;
        export where net = 0.0.0.0/0;
```

Noki, 2024-10-03:
For me switching to bird2 is fine. Can you create a PR so we can test it on more locations?

PolynomialDivision, 2024-10-12:
For me as well. However, I'm not sure if the WireGuard setup will work smoothly, since we added some 'IPC' to dynamically add interfaces to it.

spolack, 2024-10-29:
> For me as well. However, I'm not sure if the WireGuard setup will work smoothly, since we added some 'IPC' to dynamically add interfaces to it.

This might need to be patched out, but fortunately bird2 instrumentation will be super simple, as it supports wildcards on its routing protocols :)

spolack, 2024-11-08:
It's implemented and widely deployed across our network.

closed 2024-11-08 by spolack