freifunk-berlin/bbb-configs #1031: Low-Mem devices: zram-swap does not work anymore

Noki, 2024-10-25:
Low memory devices should load zram-swap:

https://github.com/freifunk-berlin/bbb-configs/blob/50698873b8566bdb399db7649f046682221461dd/roles/cfg_openwrt/tasks/conditional_packages.yml#L36-L41

This however results in the following error:

```
Thu Oct 24 00:51:16 2024 daemon.crit zram_start: kernel doesn't support swap
Thu Oct 24 00:51:16 2024 daemon.notice procd: /etc/rc.d/S15zram: zram_start: kernel doesn't support swap
```

We should either stop loading zram-swap or make the kernel support it.

spolack, 2024-10-30:
Probably related toll https://github.com/openwrt/openwrt/pull/15951

spolack, 2024-10-30:
I'd propose to either not install zram-swap on small_flash devices, which boils down ath79-tiny targets, another alternative would be to get a fix upstream. IMO saving ram is more important than saving a few bytes in the flash. Last resort would be to drop support for those and remove them from our network.

```
./model_ubnt_bullet_m2_ar7241.yml:target: ath79/tiny
./model_ubnt_bullet_m5_ar7241.yml:target: ath79/tiny
./model_ubnt_nanostation_loco_m2_xm.yml:target: ath79/tiny
./model_ubnt_nanostation_loco_m5_xm.yml:target: ath79/tiny
./model_ubnt_nanostation_loco_m5_xw.yml:target: ath79/tiny
./model_ubnt_nanostation_m2_xm.yml:target: ath79/tiny
./model_ubnt_nanostation_m5_xm.yml:target: ath79/tiny
```