freifunk-berlin/bbb-configs #1045: Core router breaking dhcp of uplink

noxilixon, 2024-11-02:
## TLDR

On the `newyorck` location we encountered multiple times that the core router lost connection to the telkom router that is used as uplink. In these situations not only the core, but also other devices in the network weren't able to connect via dhcp to the uplink anymore. Reboots did not fix it. We needed to do a full reset of the telekom router. So it is a problem on the telekom router, but seems to be triggered by the freifunk setup, as before there were no such problems. Telekom support did not help, but provided us with a newer version of their router. This didn't fix it, the problems still happen on the current model `SpeedportSmart 4`.
For now we put an edgerouter with a static ip between the core and the uplink. So this should work, but is not a solution to the problem.

noxilixon, 2024-11-02:
## Details on the last incident

<details>

<summary>The `newyorck` network went offline (no connection to the internet possible). I connected to the core router and got the logs. Apparently there was a restart of the freifunk core router 4 minutes earlier.
</summary>

```
BusyBox v1.36.1 (2024-09-28 16:33:21 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05-SNAPSHOT, r24113-fa0a7e00e8
 -----------------------------------------------------
 Host.............................: newyorck-core.olsr
 IP-Address.......................: 10.31.160.1/25
 Uptime...........................: 4 min
 Free flash.......................: 202.5M
 Average load (1m, 5m, 15m).......: 0.03 0.13 0.07
 DHCP-Clients.....................: 26

root@newyorck-core:~# logread 
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel3 < interface switch0.50 goes up
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel3 < interface ts_wg0 goes up
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel3 < interface ts_wg1 goes up
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel3 < interface eth0 goes up
Mon Oct  7 14:36:30 2024 daemon.debug bird: pipe1.pri: State changed to UP/READY
Mon Oct  7 14:36:30 2024 daemon.debug bird: pipe1.sec: State changed to UP/READY
Mon Oct  7 14:36:30 2024 daemon.debug bird: pipe2.pri: State changed to UP/READY
Mon Oct  7 14:36:30 2024 daemon.debug bird: pipe2.sec: State changed to UP/READY
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel3: Scanning routing table
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel3: Pruning table babel_default
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel1: Scanning routing table
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel1: Pruning table babel_src
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel1: Pruning inherited routes
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel2: Scanning routing table
Mon Oct  7 14:36:30 2024 daemon.debug bird: kernel2: Pruning table babel_ff
Mon Oct  7 14:36:30 2024 daemon.warn odhcpd[1568]: No default route present, overriding ra_lifetime!
Mon Oct  7 14:36:30 2024 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/network
Mon Oct  7 14:36:31 2024 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/wireless
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: started, version 2.90 cachesize 10000
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: DNS service limited to local subnets
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-nftset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 10.31.247.2 -- 10.31.247.126, lease time 5m
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 10.31.160.2 -- 10.31.160.126, lease time 5m
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using nameserver 2620:fe::10#53
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using nameserver 2620:fe::fe:10#53
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using nameserver 9.9.9.10#53
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using nameserver 149.112.112.10#53
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for test
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for local
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for olsr
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Mon Oct  7 14:36:31 2024 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 25 names
Mon Oct  7 14:36:31 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Mon Oct  7 14:36:31 2024 user.notice ucitrack: Setting up /etc/config/luci-splash reload dependency on /etc/config/firewall
Mon Oct  7 14:36:31 2024 user.notice ucitrack: Setting up /etc/config/qos reload dependency on /etc/config/firewall
Mon Oct  7 14:36:31 2024 user.notice ucitrack: Setting up /etc/config/miniupnpd reload dependency on /etc/config/firewall
[... 594 lines cut ...]
Mon Oct  7 14:40:44 2024 daemon.warn odhcpd[1568]: No default route present, overriding ra_lifetime!
Mon Oct  7 14:40:45 2024 daemon.err olsrd[2223]: OLSR: sendto IPv4 'Required key not available' on interface ts_wg0
[... cut ...]
Mon Oct  7 14:40:53 2024 daemon.err olsrd[2223]: OLSR: sendto IPv4 'Required key not available' on interface ts_wg1
Mon Oct  7 14:40:54 2024 daemon.err babeld[2444]: send: Required key not available
Mon Oct  7 14:40:55 2024 daemon.err olsrd[2223]: OLSR: sendto IPv4 'Required key not available' on interface ts_wg1
Mon Oct  7 14:40:56 2024 daemon.err olsrd[2223]: OLSR: sendto IPv4 'Required key not available' on interface ts_wg0
Mon Oct  7 14:40:57 2024 daemon.err babeld[2444]: send: Required key not available
  
```

</details>

noxilixon, 2024-11-02:
<details>
<summary>The ips and interfaces of the core router</summary>

```
root@newyorck-core:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 2001:bf7:830:a900::/56 scope global 
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c0 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::d221:f9ff:fe46:5ac0/64 scope link 
       valid_lft forever preferred_lft forever
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master switch0 state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c0 brd ff:ff:ff:ff:ff:ff
4: eth1@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch0 state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c1 brd ff:ff:ff:ff:ff:ff
5: eth2@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch0 state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c2 brd ff:ff:ff:ff:ff:ff
6: eth3@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch0 state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c3 brd ff:ff:ff:ff:ff:ff
7: eth4@dsa: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master switch0 state LOWERLAYERDOWN group default qlen 1000
    link/ether d0:21:f9:46:5a:c4 brd ff:ff:ff:ff:ff:ff
8: switch0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c0 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::d221:f9ff:fe46:5ac0/64 scope link 
       valid_lft forever preferred_lft forever
9: switch0.40@switch0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 02:00:00:00:00:01 brd ff:ff:ff:ff:ff:ff
    inet 10.31.160.1/25 brd 10.31.160.127 scope global switch0.40
       valid_lft forever preferred_lft forever
    inet6 2001:bf7:830:a928::1/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::ff:fe00:1/64 scope link 
       valid_lft forever preferred_lft forever
10: switch0.41@switch0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c0 brd ff:ff:ff:ff:ff:ff
    inet 10.31.247.1/25 brd 10.31.247.127 scope global switch0.41
       valid_lft forever preferred_lft forever
    inet6 2001:bf7:830:a929::1/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::d221:f9ff:fe46:5ac0/64 scope link 
       valid_lft forever preferred_lft forever
11: switch0.42@switch0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c0 brd ff:ff:ff:ff:ff:ff
    inet 10.31.155.225/27 brd 10.31.155.255 scope global switch0.42
       valid_lft forever preferred_lft forever
    inet6 2001:bf7:830:a900::1/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::d221:f9ff:fe46:5ac0/64 scope link 
       valid_lft forever preferred_lft forever
12: br-uplink: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c0 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::d221:f9ff:fe46:5ac0/64 scope link 
       valid_lft forever preferred_lft forever
13: switch0.50@switch0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-uplink state UP group default qlen 1000
    link/ether d0:21:f9:46:5a:c0 brd ff:ff:ff:ff:ff:ff
17: ts_wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.31.155.192/32 scope global ts_wg0
       valid_lft forever preferred_lft forever
    inet6 fe80::2/64 scope link 
       valid_lft forever preferred_lft forever
18: ts_wg1: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.31.155.193/32 scope global ts_wg1
[... 2302 lines cut ...]
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.135/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.138/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.139/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.154/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.155/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.162/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.163/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.174/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.175/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.178/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.7.179/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.50.37/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.50.38/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.50.41/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.50.42/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.131.0/27: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.31.131.64/26: seen
[... cut ...]
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.36.2.120/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.36.7.0/24: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: 10.36.7.1/32: seen
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel2: Pruning table babel_ff
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel3: Scanning routing table
Mon Oct  7 15:17:30 2024 daemon.debug bird: kernel3: Pruning table babel_default
Mon Oct  7 15:17:31 2024 daemon.debug bird: kernel1: 10.31.16

LipkeGu, 2024-12-28:
Can you show your configuration for br-uplink from /etc/config/network?

noxilixon, 2025-01-07:
```
config interface 'uplink'
	option device 'br-uplink'
	option proto 'none'

config device
	option name 'br-uplink'
	option type 'bridge'
	list ports 'switch0.50'

```

pktpls, 2025-12-17:
I think this is fixed now?

noxilixon, 2025-12-18:
We did not have those problems anymore at the `newyorck` location. But we also changed the uplink device at some point. Since there are no other reports of this problem, I would also assume that this issue is not relevant anymore.

closed 2025-12-18 by noxilixon